Ultimaker/Cura #6902: Missing Layers with Dynamic Profile

milmber, 2020-01-02:
**Application version**
Cura 

**Platform**
Mac OS 10.15.2

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Loaded model from thingyverse (STL models attached)
2. Selected dynamic print quality profile

**Screenshot(s)**
See images comparing the dynamic profile (on the left) and the standard profile (on the right)
[image]
[image]

Missing layers do not appear in print preview.

[image]

**Actual results**
Missing Layers in print when using dynamic profile.

**Expected results**
There should be no missing layers when using the dynamic profile

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[bodyForward.stl.zip](https://github.com/Ultimaker/Cura/files/4016268/bodyForward.stl.zip)
[bodyRear.stl.zip](https://github.com/Ultimaker/Cura/files/4016269/bodyRear.stl.zip)
[Rocinante_missing_layers.3mf.zip](https://github.com/Ultimaker/Cura/files/4016275/Rocinante_missing_layers.3mf.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/4016289/cura.log.zip)

**Additional information**
(Extra information relevant to the issue.)
N/A

rburema, 2020-01-03:
I can see something happening to the layer thickness when sliced with that profile. Slice the model in with the 'Dynamic' profile and set the 'Color Scheme' of the preview to 'Layer Thickness' you'll see what I mean.

rburema, 2020-01-03:
I've been informed by colleagues that one of the things the 'Dynamic' profile does is put 'Use Adaptive Layers' to true.

As a workaround, you could disable that. (Unless that was the _point_ of switching to that profile of course...)

Ghostkeeper, 2020-01-03:
Looks like a bug in Adaptive Layers thickness calculation to me.

rburema, 2020-01-07:
I was just investigating this issue a bit further, but it seems the models aren't watertight. (You can check this with the 'Mesh Tools' plugin from the marketplace).

In general, we only support slicing of [watertight models](https://knowledge.autodesk.com/support/moldflow-insight/learn-explore/caas/CloudHelp/cloudhelp/2017/ENU/MoldflowInsight/files/GUID-C0DF9415-33E2-4324-9278-61DB76CA3402-htm.html).

closed 2020-01-07 by rburema