Ultimaker/Cura #6903: "v" in front of all 4.4.1 tags? By purpose?

thopiekar, 2020-01-02:
Normally, we never had "v" in front of the version tag. Is it by purpose?

thopiekar, 2020-01-02:
Closing again since it applies for all repos: https://github.com/Ultimaker/Cura/issues/6903

rburema, 2020-01-03:
I think I started that with the 4.4-BETA and 4.4 releases (here on githib), because it's closer to the pattern that was recommended by github (text right next to the field when I had to type in the release tag). It wasn't a conscious decision by the team. Whomever released next must've picked up on that and did it for all of the 4.4.1 tags.

On the one hand I think it's a good think to be more consistent with whatever's recommended, on the other hand I shouldn't just have gone ahead and changed it by myself without notifying anyone. (In my defence, I was a little bit preoccupied with getting 4.4.0 out the door as the people who'd normally know how to do that where all out of office one way or the other for a few weeks at least :-) ).

Ghostkeeper, 2020-01-03:
Yeah I think what happened is that it was mistakenly tagged this way in Cura when 4.4.1 was released, and I copied that mistake to all of the repositories.

I've corrected the tags now. If you try the `v4.4.1` tag you'll not get it any more (but your local copy does, hurray for distributed version control). If you get the `4.4.1` tag from the old convention you'll get the same state as what `v4.4.1` used to be.

closed 2020-01-03 by Ghostkeeper