Ultimaker/Cura #6904: "Restore window position on start" don't work fine

GooodFr, 2020-01-02:
Hello !

I started to talk about it in asking a new feature : 
#6887 
I closed it and don't know how to reopen it in bug report.

On Windows 10 : 
When we open Cura, the windows seem in "full screen" but is in "reduced window".
On the "General" pane of the preferences, there's an option "Restore window position on start".
This option don't work. To work, I have to do "Ctrl and close the window".
But today, I open Cura, and it still be in "reduced window" but full screen resized.
So, no solution to Restore window position on start.

So boring to reduce this windows : 
   1. Reflex is to flip at in "reduced window", but that flip on "full screen"
   2. Flip again to flip in "reduced window"
   3. Still can't change the size, so move the window to a wrong place
   4. Resize smaller to have a window able to fit in the screen
   5. Move again to the correct place the window
   6. Finally resize the window to the desired size

Ragards,

Ghostkeeper, 2020-01-03:
I'm not able to reproduce this problem on Ubuntu 19.10. If I close Cura while the screen is maximised, it remains maximised the next time I open it. If I close Cura while the screen has a customised size (not maximised) then the same size is restored the next time I open it. No Ctrl key is involved.

Maybe only on Windows?

fieldOfView, 2020-01-03:
What I can confirm (on Windows 10) is the following:
* load Cura and set the window to a custom size and position
* make sure the "Restore window position on start" option is checked
* restart Cura
-> windows is correctly placed in the custom position and size

* now maximize the window
* restart Cura
-> window is (correctly) maximized on startup

* now restore the window to not maximized
-> window (incorrectly!) does not restore to previous custom position and size

It looks like this is a regression, because I distinctly remember having implemented it so that the stored window size and position is not changed when the window is maximized.

I am not entirely sure if this is what @GooodFr is describing because his terminology confuses me. The window can be resized and repositioned when it is in its non-maximised screen-filling state.

Ghostkeeper, 2020-01-03:
That indeed reproduces and seems like a bug to me.

GooodFr, 2020-01-04:
No ! You right @fieldOfView ! It's what I have !
I didn't noticed that I close almost all time Cura when it's maximized window.
So, it memorize maximized size for the no-maximised size.
That why I so often have no-maximised windows in full screen.

"Ctrl and close the window" don't especially work, it was because I closed Cura in no-maximised window.

Sorry to confuse you, my English is so bad :)

fieldOfView, 2020-01-06:
I have made a pull request to fix this issue here: https://github.com/Ultimaker/Uranium/pull/561

Ghostkeeper, 2020-01-16:
Thanks fieldOfView. That indeed fixes this issue.

closed 2020-01-16 by Ghostkeeper