Ultimaker/Cura #6905: Slicing won't start on 4.4

Podaros, 2020-01-02:
Cura version 4.4, also tried older versions, screenshot is from 3.6 (same problem)

Windows 10, ryzen 5

Ender3 Pro

[image]

Nothing follows after that, no crush, no error.

[cura.log](https://github.com/Ultimaker/Cura/files/4017483/cura.log)

Can't actually find a solution in any other post so I made my own.

rburema, 2020-01-03:
Hm, there's this in the logs:

> 2020-01-02 20:45:34,064 - DEBUG - [Thread-18] UM.Backend.Backend._backendLog [94]: [Backend] [ERROR] Could not connect to the given address

Could you check your firewall settings? We use a localhost port to connect to the engine (the part that does the slicing). Some firewalls also block/apply their rules to messages that go both from and to the same machine.

Podaros, 2020-01-03:
Yeah, I've tried turning off the firewall but this didn't change anything.

Ghostkeeper, 2020-01-03:
It does actually start the engine though (receiving the license and such). I think it may have something to do with that KeyError in the log. Did you try 4.4.1 or is it still 4.4.0?

rburema, 2020-01-03:
Just in case it _is_ the firewall: Did you also restart Cura when the firewall was off?

(P.S. You shouldn't have to turn off the entire firewall, though.)

rburema, 2020-01-03:
The KeyError log also occurs in 3.6, so I doubt a bit it's something we introduced in 4.4 (and thus fixed with 4.4.1)

rburema, 2020-01-03:
Seems like it adds the extruder _after_ an attempt to fetch it:

> 2020-01-02 23:35:50,252 - DEBUG - [MainThread] UM.Logger.logException [84]: Exception: An exception occurred in inherit function =extruderValue(adhesion_extruder_nr, 'adhesion_type'): '0'
2020-01-02 23:35:50,255 - DEBUG - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2020-01-02 23:35:50,257 - DEBUG - [MainThread] UM.Logger.logException [88]:   File "X:\3.6\build\inst\lib\python3.5\site-packages\UM\Settings\SettingFunction.py", line 95, in __call__
2020-01-02 23:35:50,262 - DEBUG - [MainThread] UM.Logger.logException [88]:   File "<UM.Settings.SettingFunction (0x1f3a0acfc88) =extruderValue(adhesion_extruder_nr, 'adhesion_type') >", line 1, in <module>
2020-01-02 23:35:50,267 - DEBUG - [MainThread] UM.Logger.logException [88]:   File "X:\3.6\build\inst\lib\python3.5\site-packages\cura\Settings\CuraFormulaFunctions.py", line 41, in getValueInExtruder
2020-01-02 23:35:50,273 - DEBUG - [MainThread] UM.Logger.logException [88]: KeyError: '0'
2020-01-02 23:35:50,281 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [90]: Extruder[creality_ender3_extruder_0 #2] added to [Creality Ender-3] at position [0]

Podaros, 2020-01-03:
Just updated to 4.4.1 to check, same result.
Yes I did restart Cura after turning off the firewall.

Here's the new log from 4.4.1
[cura.log](https://github.com/Ultimaker/Cura/files/4018532/cura.log)

rburema, 2020-01-03:
Hm, it seems to inherit a wrong printer config from the previous version(s):

> 2020-01-03 10:41:03,727 - ERROR - [MainThread] UM.Settings.ContainerStack.deserialize [390]: When trying to deserialize Creality Ender-3, we received an unknown container ID (Creality Ender-3_user)
2020-01-03 10:41:03,730 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Error in the format of container Creality Ender-3: When trying to deserialize Creality Ender-3, we received an unknown container ID (Creality Ender-3_user)
...  {... snip most of the traceback ...}
2020-01-03 10:41:03,749 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerStack.py", line 391, in deserialize
2020-01-03 10:41:03,753 - ERROR - [MainThread] UM.Logger.logException [88]: UM.Settings.ContainerFormatError.ContainerFormatError: When trying to deserialize Creality Ender-3, we received an unknown container ID (Creality Ender-3_user)

Could you try adding a new printer from scratch?

Podaros, 2020-01-03:
Yeah it did give me an "error" and asked to reset my settings, and I did, then added my printer again, I tried an Ultimaker printer as well, didn't change anything.

Podaros, 2020-01-04:
Update: I reinstalled it, downloaded directly a 4.4.1
Nothing changed, tried fiddeling with the model and slicing it afterwards, added a different printer, restarted my pc and did it all again still nothing.
Should I wait for a newer version? Are there any coding ways to fix it?
Screenshot for better idea of what I see when I'm trying to slice.

[image]

rburema, 2020-01-07:
It's probably an issue with the configuration folder, given the `we received an unknown container ID (Creality Ender-3_user)`. We could have a look in there, but it may contain user information. They're in the same folder the logs are in (more specifically, the folder the logs are in _is_ the configuration folder). We'd need to look at where the value 'Creality Ender-3_user' comes from.

Podaros, 2020-01-08:
You mean all of them folders and files?

Ghostkeeper, 2020-01-13:
Yeah, or more specifically, all of the files related to the Creality Ender 3. We're mostly interested in the `machine_instances` and `user` folders, but to reproduce your issue from your configuration folder it's best to have the entire configuration folder.

Your reproduce steps so far don't reproduce the issue for us. We've added an Ender 3 and loaded a model and it just slices fine. So we're curious why it's not finding the user settings stack for you - maybe there is a parse error or something. And after that we're curious how you got the configuration in that state, which will be more difficult to figure out.

Podaros, 2020-01-20:
Can't upload this file type here for some reason.

mahtDFR, 2020-01-20:
If you zip the files github will allow you to attach them here.

Podaros, 2020-01-20:
[cura.zip](https://github.com/Ultimaker/Cura/files/4086083/cura.zip)

mahtDFR, 2020-01-20:
Just tried with your exact configuration and I'm still unable to reproduce the issue. It slices just fine for me, and doesn't throw up errors mentioned in the logs quoted above.

Uploaded configuration I used and logs generated here for reference
[4.4.1-20-01-20.zip](https://github.com/Ultimaker/Cura/files/4086245/4.4.1-20-01-20.zip)

Podaros, 2020-01-20:
Maybe it's about my windows? Is reinstalling windows worth a try?

mahtDFR, 2020-01-20:
> Maybe it's about my windows? Is reinstalling windows worth a try?

I think that would be a rather drastic step

Ghostkeeper, 2020-01-22:
The engine is being started with the correct socket number, but it's not able to connect back for some reason. Perhaps a VPN client?

xfoxdragonx, 2020-01-22:
i got the same thing , i got a flashforge printer and made a custom printer profile , it will not slice , but when i select ultimake it will also not slice so i don't think it have any to do with printer profile

rburema, 2020-01-23:
@xfoxdragonx Could you share your logs? In the meantime, you can look at whether your firewall blocks localhost ports.

xfoxdragonx, 2020-01-23:
[cura.log](https://github.com/Ultimaker/Cura/files/4101965/cura.log)

rburema, 2020-01-23:
@xfoxdragonx It seems you are missing the engine (backend) binary:

> UM.Logger.logException [84]: Exception: Unable to find backend executable: C:\Program Files\Ultimaker Cura 4.4\CuraEngine.exe

Can you check if it's there? If so, it's probably a permissions problem. Otherwise, you might have to reinstall Cura.

xfoxdragonx, 2020-01-23:
[image]

rburema, 2020-01-23:
You specifically need `CuraEngine.exe` to be in that folder, it should've been installed with the rest of the program.

xfoxdragonx, 2020-01-23:
i already reinstalled cura 
[image]

xfoxdragonx, 2020-01-23:
glasswire also see it's trying to do some ( it's not blocking it )

[image]

xfoxdragonx, 2020-01-23:
to try and fix it i deinstalled it , and reinstalled version 4.4.1, 
than i deinstalled it and installed 4.1.0 also don't work ,
than i reinstalled 4.4.1 again , try to open the engine with runn as admin ( befor slicing and during slicing )

xfoxdragonx, 2020-01-23:
[image]

xfoxdragonx, 2020-01-23:
strange thing i don't see the slicer engine on glass wire so that mean's it has not try to use anny connection yet ( so i don1't think that it is starting )

rburema, 2020-01-23:
> so i don1't think that it is starting

Well that's the problem the logs point out. The front-end acts as if it can't find that file, but I can see in your screenshot that it _is_ there. I suspect there may be something else wrong with the command we try to execute rather than the program being executed (like the parameters). The logs unfortunately do not show the entire command.

xfoxdragonx, 2020-01-23:
are there other log file's i need to send ?

xfoxdragonx, 2020-01-23:
i run the curacli as admin i get this ( it also starts cura again stll will not slice )

[image]

xfoxdragonx, 2020-01-23:
i will install it on a difrent hdd , windows is iffy at giving acces to c drive sometimes

xfoxdragonx, 2020-01-23:
still no luck , (i selected the ultimaker s5 as printer just so it has a printer it should work with )

xfoxdragonx, 2020-01-23:
2020-01-23 10:11:18,363 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Unable to find backend executable: F:\program files\Ultimaker Cura 4.4\CuraEngine.exe

the file is there and not on c drive so windows would not be as likely to block it when the windows for adding to firewall poped up i gave premisions for all networks

xfoxdragonx, 2020-01-23:
ERROR - [MainThread] UM.Logger.logException [84]: Exception: Unable to find backend executable: F:\program files\Ultimaker Cura 4.4\CuraEngine.exe

here i try to runn the slicer engine as admin and it spits out the folowing

ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 126, in _runEngineProcess
ERROR - [MainThread] UM.Logger.logException [88]:   File "subprocess.py", line 640, in __init__
ERROR - [MainThread] UM.Logger.logException [88]:   File "subprocess.py", line 859, in _get_handles
 [MainThread] UM.Logger.logException [88]:   File "subprocess.py", line 735, in _get_devnull
ERROR - [MainThread] UM.Logger.logException [88]: FileNotFoundError: [Errno 2] No such file or directory: 'nul'
 DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [429]: Sending slice message took 0.6013731956481934 seconds
DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles

xfoxdragonx, 2020-01-23:
after canceling the slice job it gif me 

2020-01-23 10:53:31,942 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2020-01-23 10:53:31,945 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 126, in _runEngineProcess
2020-01-23 10:53:31,947 - ERROR - [MainThread] UM.Logger.logException [88]:   File "subprocess.py", line 640, in __init__
2020-01-23 10:53:31,950 - ERROR - [MainThread] UM.Logger.logException [88]:   File "subprocess.py", line 859, in _get_handles
2020-01-23 10:53:31,952 - ERROR - [MainThread] UM.Logger.logException [88]:   File "subprocess.py", line 735, in _get_devnull
2020-01-23 10:53:31,955 - ERROR - [MainThread] UM.Logger.logException [88]: FileNotFoundError: [Errno 2] No such file or directory: 'nul'

wondering what this is aboud 
File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py"
x is a drive , but a network drive on my server ,

fieldOfView, 2020-01-23:
The reference to X: is an artifact of the build system. Cura gets packaged from an X: folder on a build computer, and a reference to that path gets backed into the built version. Just to rule out your X: drive as the cause of the issue, could you try disconnecting that drive and slicing?

fieldOfView, 2020-01-23:
> 2020-01-23 10:53:31,955 - ERROR - [MainThread] UM.Logger.logException [88]: FileNotFoundError: [Errno 2] No such file or directory: 'nul'

That sort of explains things. The issue is not that curaengine.exe cannot be cound, but /dev/nul cannot be found. The next question is why subprocess.py tries to access /dev/nul on a Windows system in the first place.

xfoxdragonx, 2020-01-23:
removing the x network drive and also unplugging my network cable don't make a difrence

Ghostkeeper, 2020-01-24:
The stack trace is a bit weird to me. It's basically these lines:

1. Cura starts CuraEngine with this call:
    ```
    subprocess.Popen(command_list, stdin = subprocess.DEVNULL, stdout = subprocess.PIPE, stderr = subprocess.PIPE, **kwargs)
    ```
2. In the subprocess module of Python there's the `Popen` constructor calling this:
    ```
    (p2cread, p2cwrite,
         c2pread, c2pwrite,
         errread, errwrite) = self._get_handles(stdin, stdout, stderr)
    ```
3. The `_get_handles` function checks `stdin`, `stdout` and `stderr` what type of input it is. If either of them is `subprocess.DEVNULL`, then it calls this one:
    ```
    p2cread = msvcrt.get_osfhandle(self._get_devnull())
    ```
    In our case that happens with `stdin`, since we passed `subprocess.DEVNULL` to it in the `Backend` class.
4. `_def_getnull` simply returns `os.devnull` which is set to `"nul"` for Windows.

Looking at the documentation and examples for usage of `os.devnull` it should be reliable for programmers to `open(os.devnull)` to read from or write to the null device. So I don't see how this could give an error.

As a workaround we can try replacing that with `None`:

    return subprocess.Popen(command_list, stdin = None, stdout = subprocess.PIPE, stderr = subprocess.PIPE, **kwargs)

`None` is the default and it behaves differently according to the code. It creates an input to the standard-in pipe. However since we don't send anything from that pipe it'll not get any data from that so it should behave the same. For me it still works if I change that.

xfoxdragonx, 2020-01-24:
how do i change it ?

Ghostkeeper, 2020-01-24:
You'd need to change that in Uranium, which requires re-packaging Cura for Windows. You basically need a development environment for Cura to change it. I aimed that more at the developers here.

xfoxdragonx, 2020-01-24:
:D y i don't realy know anything of coding :D

LipuFei, 2020-01-27:
Changed `stdin` from `os.DEVNULL` to `None` in https://github.com/Ultimaker/Uranium/commit/05a7455a34b027e0dccafdbeac2c3bfcea0794c4

xfoxdragonx, 2020-01-27:
can i test ?

fieldOfView, 2020-01-27:
You can, but only in a somewhat hacky way for now. You have to know your way around a zip file and must be prepared to reinstall Cura if it does not work.

* Download this file: https://raw.githubusercontent.com/Ultimaker/Uranium/05a7455a34b027e0dccafdbeac2c3bfcea0794c4/UM/Backend/Backend.py
* Navigate to where you have Cura installed (Eg `C:\Program Files\Ultimaker Cura 4.4\`)
* Open the file `python35.zip` from that folder in your favorite zip application (eg 7Zip)
* Navigate to `UM\Backend`
* From that zipped folder, remove the file named `Backend.pyc`
* In that same zipped folder, place the file you downloaded. Note that that is `Backand.py`. Do not change the extension, keep it at .py

The file you removed is a "compiled" version of the python file (hence the 'c' for 'compiled' in the extension). You replace it with an non-compiled fixed version.

Now start Cura and try. If it works, then great! If not, then please post another log. It might be that the fix does not work, or that this hacky way of patching Cura went wrong.

xfoxdragonx, 2020-01-27:
that zip seems to be protectet , i could remove the file but not extract or put the new file in

xfoxdragonx, 2020-01-27:
never mind was just windows protecting c drive

xfoxdragonx, 2020-01-27:
> You can, but only in a somewhat hacky way for now. You have to know your way around a zip file and must be prepared to reinstall Cura if it does not work.
>
>     * Download this file: [...]

thank you so mutch all it finaly works :D this was the solution ( dono wat it is but it works )

Ghostkeeper, 2020-01-27:
Thanks for testing!

@LipuFei we'll probably need to have a ticket for this in order to have the QA team test this as well, no?

LipuFei, 2020-01-27:
> Thanks for testing!
> 
> @LipuFei we'll probably need to have a ticket for this in order to have the QA team test this as well, no?

Yes, the JIRA ticket is CURA-7142

xfoxdragonx, 2020-01-27:
will this be implemented in the next update or should i not update for a while ?

Ghostkeeper, 2020-01-27:
It'll be in the next release (the 4.5 beta, and then 4.5 stable). If you install 4.5 it should work in that version without needing to make local modifications.

NGTPadre, 2020-03-21:
> You can, but only in a somewhat hacky way for now. You have to know your way around a zip file and must be prepared to reinstall Cura if it does not work.
>
> * Download this file: [...]

hey man, I have the same issue but this link ain't working anymore. Could you send a new one?

fieldOfView, 2020-03-21:
The link works for me.

NGTPadre, 2020-03-21:
> The link works for me.

I am kinda new to github and I don't really know if what I see is what I should see but when I open the link I see some cmd-like stuff
[image]

fieldOfView, 2020-03-21:
That is correct. Download that file, and follow the rest of the steps.

[3 more replies]

closed 2024-12-07 by github-actions[bot]