Ultimaker/Cura #6908: [master] some setting values can't be changed

smartavionics, 2020-01-03:
**Application version**
master branch

**Platform**
Windows & Linux

**Printer**
Anycubic i3 mega.

**Reproduction steps**
1. Load the attached project
2. Try to change bed temperature or build plate adhesion type

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Setting value "jumps back" to previous value, entered value is ignored.

**Expected results**
Settings should retain the new values

**Project file**
[Cannot.change.values.zip](https://github.com/Ultimaker/Cura/files/4018546/Cannot.change.values.zip)

**Log file**

Haven't spotted anything relevant.

**Additional information**

This issue was reported for my Cura build. Apparently, the 20191214 build was OK but the 20191228 build has this problem so it's something that changed upstream that is causing this because I haven't made any changes to the settings code.

smartavionics, 2020-01-03:
Not all projects have this issue as I haven't noticed this behaviour for my own projects.

Ghostkeeper, 2020-01-03:
Yeah this reproduces with that project file.

I notice that the `user` profile sets the bed temperature to 100 degrees, rather than the 60 that's being displayed in the interface.

evtrados, 2020-01-08:
Could not reproduce the issue using several other project files. In any case, though, this needs to be addressed.
Developers, please see CURA-7095.

smartavionics, 2020-01-08:
Hi, I also have other problems with the current master branch. See https://github.com/Ultimaker/Cura/commit/3e6723f378ac9da70e5c3ddbd2e6d4d9eca45667.

Ghostkeeper, 2020-06-25:
Looks like this fixed itself. The project file displays the correct bed temperature now (100) and allows modifying it fine. I don't know which changes fixed it in the past half a year, but we've done many bugfixes so this may have been among them.

closed 2020-06-25 by Ghostkeeper