Ultimaker/Cura #6909: Cura 4.4.1 Prime Tower - skipping layers for secondary extruder

mkudzia84, 2020-01-03:
**Application version**
4.4.1

**Platform**
Windows 10/64 bit

**Printer**
Custom setup for E3D tool changer 

**Reproduction steps**
1. 4 extruder setup, using extruder 1 and 3
2. Enabled prime tower

**Screenshot(s)**
Layer 1 [https://ibb.co/G0h9mRp](url) 
Layer 2 [https://ibb.co/x7CzPG3](url)
Layer 3 [https://ibb.co/X2qMTYc](url)

**Actual results**
Secondary extruder is only primed at the layer where the tool is activated.
In this setup this is happening every 2nd layer.
This however means - it is skipping one layer and doesn't have a surface underneath - material doesn't stick; it follows underextrusion on the model. 
( check images -
layer 1 - both are primed
layer 2 - only primary extruder creates an outline, secondary extruder doesn't
layer 3 - primary extruder and secondary extruder create outline; secondary extruders 'primed material' is hanging in the air

**Expected results**
prime tower is being created on every layer .

mkudzia84, 2020-01-03:
Just finished writing a CuraEngine patch that fixes it - compiling and testing.

Block137, 2020-01-08:
How will the fix work?
Will it activate secondary extruder just to prepare the prime tower? or will it do 2 layers of priming at once?

mkudzia84, 2020-01-08:
I have already did the fix and provided it in a pull request - it's here

https://github.com/Ultimaker/CuraEngine/pull/1177

New strategy is :
- on tool change, slicer checks if previous extruder primed into the tower; if not add gcode to prime the tower with prevExtruder prior to the tool change (new behaviour)
- after the tool change, prime into the tower (current behaviour)
- at the end of layer processing, check which extruders haven't been primed in this layer; for each extruder that hasn't been primed, generate gcode for priming using currently active extruder

In this scenario - in example, 
```
layer #10, T0, T1 is active, T2, T3 is not active
- layer starts with T1 printing (T1 was last extruder in layer #9)
- T1->T0 is triggered
- [new behaviour] check if T1 primed in layer #10, it hasn't so generate gcode for priming into T1 prime tower band
- change T1->T0
- [current behaviour] check if T0 is primed, is not so generate gcode for priming into T0 prime tower band
- print with T0
...
...
- [new behaviour] end of layer processing, check which extruders haven't been primed -> [T2, T3] - for T2, T3, generate gcode for printing into the T2, T3 tower prime bands using current extruder settings/curren extruder [T0]
```
This guarantees all of the requested prime tower bands will be printed
No new tool changes are added

I have tested this in various configurations for my E3D toolchanger equiped with 4 toolheads - all was ok.

Block137, 2020-01-08:
Is it possible to also fix the issue where primary extruder would print prime tower without tool change?

In 2 extruders case, tool change happens every 2 layers and I noticed T0 would prime after every layer 
(T1_to_T0 > tower > object > new layer > tower > object > T0_to_T1)

Ghostkeeper, 2020-01-13:
This is actually intended behaviour. The prime tower consists of multiple nested shells. The outer shell is always printed, to ensure the stability of the prime tower (even when using materials that don't stick well together). For the rest of the shells, it essentially just purges in the bucket it's printed earlier.

Not purging when it's not necessary:
* saves printing time and material, and
* prevents unnecessary extruder switches when there are multiple consecutive layers printed only with one extruder, such as for this [traffic cone](https://www.thingiverse.com/thing:21773).

> This however means - it is skipping one layer and doesn't have a surface underneath - material doesn't stick; it follows underextrusion on the model.

I don't understand how this problem can happen. The layers that it skipped are the layers where it started with that extruder because it ended the previous layer with that extruder. So it basically just printed one layer with extruder 2, then continues with the next layer with extruder 2 without purging. It shouldn't have underextrusion then, just as there is no underextrusion when you print multiple layers with single-extrusion.

Why does purging in mid-air cause underextrusion after the purging?

mkudzia84, 2020-01-13:
Well, the strategy you have implemented is not adequate; first of all 'mid air extrusion' inside of the prime tower bucket has the following issues:
a) Strings everywhere (can live with but not ideal/unnecessary messy)
b) It causes underextrusions/blobs 

The reason why you want to extrude against existing layer is to - extruder material and also equalize the pressure within the nozzle/melting chamber. Extruding mid air doesn't do that - pressure in the nozzle is to low. (depends on temp and material)

Also other issues (more prevalent in 0.6+ nozzles, or 0.4 nozzles with PET/materials with higher viscosity)
a) the extruded string of material doesn't get detached from the nozzle by the edge of purge bucket, get's dragged onto the model and we end up with a mess on the model surface
b) the extruded string of material gets detached, but 'pulls' some excess material out of the nozzle - causing the underextrusion on the model.
c) messy random strings/debris within the Prime Tower (which you nerfed and made a purge bucket) touch nozzles of new tools, melt and attach to it causing - material cross contamination and blobs on the model.

As such - priming against a layer of material (not mid air) - with normal retract/z-hop is advisable here and prevents those issues.

As for 
- saves printing time and material, and
- prevents unnecessary extruder switches when there are multiple consecutive layers printed only with one extruder, such as for this traffic cone.

As discussed in the pull request:

 - extra tool changes; this is the big one as generally - this is the most time-consuming action (on my tool changer takes around 20s...)
- as outlined in the Pull Request and this thread, my strategy doesn't introduce any extra tool changes:

The prime tower band for specific tool is printed when:
a) During Tool Change operation (T0 ->T1 for example) 
    - [new behaviour] prior to the tool change, slicer checks if the prime tower band for Tool Previous (T0) has been already been primed in the current layer, if not it inserts the gcode for priming it,
    - Tool change 
    - [current behaviour] Prime the Tool New (T1) 
b) At the end of the layer processing - check which prime tower bands haven't been printed, use the currently active Tool to print them [ no extra tool change ] 

This means 
1) We have no extra tool changes
2) it's best effort of trying to print prime tower bands with the appropriate tool
3) if b) is not possible, use the active extruder at the end of the layer (no extra tool change)

Only drawback:
b) Because at the end of the layer we pring the missing prime tower bands with last active extruder it's possible that for example, we will be printing a layer band with tool equipped with 0.25mm nozzle - where that band is generated for a tool with 0.6mm nozzle.
I have tested this and this is not an issue - if the toolhead offsets are well clalibrated  (mine is down to 0.03mm x/y variance) - there is enough contact surface - stacking 0.6 on 0.25 to get layer adhesion and consistent extrusion (this was tested with PLA 0.25mm, 0.6mm petg - they don't stick well together so it was the worst case scenario)

As for extra print time and material:
yes, we waste a bit more material and add some print time but:
- on 4 extruder setup added time was 4s-6s per layer.
- on a 30cm print extra wasted material added up ~20g)

If you would ask anybody, (especially doing multi material prints) - that is a tiny overhead compared to improved reliability and print quality.

Ghostkeeper, 2020-01-14:
We've discussed this issue with the development team and consider those points to be a good reason to add this feature. However we'll request that you add a boolean setting to enable/disable the extra prime tower layer, since it won't be necessary for all towers, and we'd like to keep the default behaviour as much the same as possible. I'll repeat this message in your pull request :)

NMe84, 2020-02-28:
> This is actually intended behaviour. The prime tower consists of multiple nested shells. The outer shell is always printed, to ensure the stability of the prime tower (even when using materials that [...]

It does exactly the opposite, especially with that traffic cone. Because the first extruder's color is always used in the outer shell of the prime tower it keeps switching to that color to build up the tower, and it leaves less surface on the tower to purge than when you had the whole layer for purging a single color.

What happens with that traffic cone is that it will print just fine with only red up until the first layer that is a white band. There it will build the outer layer of the prime tower, swap tools, purge the inner layer of the prime tower with white, print the white band, move up the Z-axis, swap to red again just to print the outer layer of the prime tower again, etc. There are way more tool changes than required (because let's be honest, we could just keep the white filament loaded for the entire white band), and because white is such a "weak" color the prime tower has to be pretty big to have enough volume left to purge all the red out before being able to actually print white. I ended up printing a red traffic cone with pink bands yesterday because of all this, even though the purge tower was a lot heavier than the model and the volume of the inner column was larger than that of the outer column...

The traditional purge block where each layer of the purge block never has more than one color makes a lot more sense for dual extrusion. It means fewer tool changes and less bleeding of colors. I don't get why it's not an option to do that in those cases where your approach doesn't make sense. Or at the very least, if you want to keep your current default approach: it shouldn't matter which extruder builds the outer layer. If the model itself doesn't require a tool change, the prime tower shouldn't trigger one.

Ghostkeeper, 2020-03-04:
Yes indeed you could skip even more purging by swapping the outer shell to a different colour as well. The prime tower worked like that up until Cura 3.5. This is a different issue though, being discussed here: https://github.com/Ultimaker/Cura/issues/5282

JohnTasto, 2020-12-03:
Everyone seems to be complaining that their secondary extruder is skipping layers - for me, it is my primary extruder that skips layers. Somehow my secondary extruder is printing the outer shell, and I can't get my primary extruder to do it.

I'm printing PETG primary (T0) and PLA as support (T1), so I get a PLA prime tower. The PETG only primes every other layer, so it has nothing to stick to except the nozzle, and ends up as tumorous growths on my print. After my last print, there was only a few wisps of PETG that managed to stay in the prime tower - the rest was stuck all over my print.

Any suggestions would be appreciated. The easiest solution I can think of at this point is rewiring my printer to flip T0 and T1, then switching everything in Cura so it treats T1 as primary. I'm not even sure that'd work though.

taconiteable, 2021-01-09:
Why not just let us choose the "Main" Prime Tower Extruder. That way we as users can decide which Extruder is used for the outer shell and we can make sure that only a extruder that is used in the print is assigned to it

Ghostkeeper, 2021-01-18:
That kind of clashes with the Adhesion Tendency automatically determining the extruder to use for the outer shell. It's either automatic or manual, really. We've previously chosen the automatic approach (Adhesion Tendency) because it's more friendly to novices who don't know about selecting the best extruder to print the shell of the prime tower with. Or even what a prime tower is and why it's necessary.

wawanbreton, 2021-05-14:
Hi, any news on this subject ? I just wanted to agree that I don't like when there is no support and extruder is printing in the air. It is kind of messy, especially when the extruded has not been used since many layers. Sometimes the new filament goes upwards and sticks to the extruder, which doesn't help to keep it clean.

By the way, has it been considered to use a fixed filling pattern when no actual extruder change is requred ? That would be a compromise between no support at all, and a full layer, saving filament and time. Maybe it could even be used for the outer shell when possible, saving even more filament.

tommylux, 2021-11-01:
Any progress on this? I notice that cura is printing mid-air in the prime tower when it needs to print a new colour that it hasn't yet printed before.

flipace, 2021-11-10:
Same issue here :( Is there any way to work around that issue in Cura?

[image]

Ghostkeeper, 2021-11-17:
Do you have a project file of that issue, @flipace?

flipace, 2021-11-17:
> Do you have a project file of that issue, @flipace?

Here are two example files with issues for a config with an A10T 3-in-1-out printer. (Shared nozzle, heater)

1 - https://drive.google.com/file/d/1fKYTF41tb23aN9dgdhhtuRCkTqn_F9HJ/view?usp=sharing: Here, the 2nd extruder is used for the last few layers. Cura "primes" it mid-air

[image]

2 - https://drive.google.com/file/d/1NGeoZtOV3kdPZBvxZNygv9ay9oonxGNd/view?usp=sharing: 3rd extruder is used for the last few layers. Cura doesn't prime this one at all 🤔 
[image]

Is it some misconfiguration on my side, or am I understanding the prime tower wrong?

Block137, 2021-11-17:
1 - Prime mid-air is the problem being discussed here. The issue is still open so it is not solved yet.

2 - If not priming, check "Prime Tower Minimum Volume" first, setting it to 0 will skip priming.

But the first image you showed (floating layer outside) is new, it could be a separate bug.

tommylux, 2021-11-17:
I wish cura could use square towers so it can adjust the amount it needs to prime based on the infill for the tower. Ie for layers where no priming is required, it could use less infill. However cura primes the same amount of filament per layer.
That’s why I’ve switched to PrusaSlicer. Bit of a learning curve but the honourable gentleman may like features such as ‘paint material colour’ to paint a model with the material he wants in the new PrusaSlicer beta.

[cura-issue-1]<https://user-images.githubusercontent.com/953984/142181369-c0ed2507-76bc-43e0-a07d-bfb4b3e39ec1.gif>

[image]<https://user-images.githubusercontent.com/953984/142181109-9abe3fa8-65c9-4110-b033-23258e9f6846.png>

flipace, 2021-11-17:
> 1 - Prime mid-air is the problem being discussed here. The issue is still open so it is not solved yet.
> 
> 2 - If not priming, check "Prime Tower Minimum Volume" first, setting it to 0 will skip priming.
> 
> But the first image you showed (floating layer outside) is new, it could be a separate bug.

I see, thanks - I will try to reproduce the case from the floating outside layer. Didn't store the project :(

wawanbreton, 2021-11-17:
Hi @flipace, what is the behavior you would expect ? Do you think every previous layer should be filled with the same filament, or use the actual filament, in order to save time by not constantly switching nozzles ?
@tommylux I was actually thinking about replacing the circles by spirals : that would (slightly) reduce the priming time, and also remove the shocks at each turn. And finally, we could have less dense spirals printed at layers where priming is actually useless, thus having a support but with less filament. What do you all think about this ?

tommylux, 2021-11-17:
How are you producing those gifs?

To the topic at hand. consider how prusaslicer handles the task. For layers where they are the same colour, it builds a shell.
[image]

Where it needs to change colour it wipes in infill to save filament and then bridges over the tower for additional wiping
[image]

This way, it can adjust the amount of support it needs to purge by bridging over the tower shell. You cannot achieve this with cura purge tower using concentric infill as there is nothing to bridge to and doesn't wipe as well.

Regarding Cura's design, yes, every layer should be printed. If no colour change, it should at least continue building the tower. You need to ideally print onto something, not onto nothing. Why would we need circules for every extruder? surely you can purge ontop of the last colour.

As a side note, you can change how much to purge based on colour change:
[image]
I think Prusa is winning in this category.

Block137, 2021-11-18:
You have to consider the case where there are different materials that don't bond to each other. Using Prusa solution in this case will create a tower that delaminate. But in case of same material, it will be a lot more efficient. 
The solution is leaning toward implementing 2nd type of tower (the original "bucket" tower and the "full" tower).

flipace, 2021-11-18:
> How are you producing those gifs?
I'm using https://giphy.com/apps/giphycapture, but there's also Kap which I also sometimes use and is slightly more flexible (https://getkap.co/)

> what is the behavior you would expect?

Honestly, anything that ensures that the layers are not printed mid-air, since this is at the moment not really working at all 😅  IMHO wasting material here is better than not having the option to use the prime tower at all.
I'm afraid I'm not too deep into the whole topic to offer more helpful suggestions (especially taking into account the many different combinations of configurations which you have to consider) :(

tommylux, 2021-11-18:
True, but generally with colour printing, you want to be using the same material anyway because you are either doing colour mixing or multi-part colour prints, both would require good adhesion. Not sure what application you would have to print two separate models with different materials, because you’ll just be wasting filament by having a purge tower. Isn’t the best solution one which applies to the majority?

Block137, 2021-11-18:
Most common use is separable support. PVA (water soluble) for example doesn't bond well to most other materials, is soft and would make a very unstable tower if printed in alternating layers.

flipace, 2021-11-18:
> Most common use is separable support. PVA (water soluble) for example doesn't bond well to most other materials, is soft and would make a very unstable tower if printed in alternating layers.

would it be possible to have separate towers per nozzle maybe? that could then be a "simple" checkbox i guess and there wouldn't be a need for crazy configuration of a single tower?

wawanbreton, 2021-11-18:
Well, I personally think that a single tower is better, because it saves space on the bed, and it is also more consistent when getting very high : having multiple towers would make them very small (or use a lot of space) and they may unstick as the bed is moving. However, having multple materials seems to require that all the rings are printed with the proper one. But having them all around each other can only be better for global stability.

NMe84, 2021-11-18:
> Most common use is separable support.

Do you have a source for that claim? Because I very much doubt it's the single most common use.

Block137, 2021-11-18:
> 
> Do you have a source for that claim? Because I very much doubt it's the single most common use.

I wasn't clear, I meant most common use for "different material" per extruder.
The most common use for multi-extruder as a whole is probably multi-color of same material.

wawanbreton, 2021-11-20:
So, if I understand correctly, in the worst case we can't print different materials on top of each other. That means each ring should be consistent. And if we want each ring to be printed, that means we have to print all the layers, and then change tools every layer. The final behavior would be very simple :

- L0 => prime E0 => print E0 part (if any) => change E0->E1 => prime E1 => print E1 part (if any)
- L1 => prime E1 => print E1 part (if any) => change E1->E0 => prime E0 => print E0 part (if any)
- L2 => prime E0 => print E0 part (if any) => change E0->E1 => prime E1 => print E1 part (if any)

That would work for every situation, and the tower would be very strong and consistent. We still have only one tool change for every layer, but this behavior is much more filament-consuming.

The only situation where we can save some is when re-priming the same extruder, but only for the inside rings because we want the outside to be strong. At the end of the day, the behavior would be very similar to the current one, except that we can print "missing" layers with a non-100% filling.

So what I propose is to :
- Add an infill density parameter for useless layers, which would default to 0 to keep the actual behavior
- Replace circle by spirals to save time and allow for partial infill when possible

Or, if we want to get closer to the PrusaSlider behavior, we could print a shell with walls to contain many towers. It may save a lot of filament and time, but it seems to break the rule of multi-materials on top of each other. In this situation, we would still have to print an infill with the proper material, and change tools at every layer...

Block137, 2021-11-21:
I can think of at least 4 ways to print a prime tower, each for different use case.

1. Laying right on top of each other, no separate ring, for multicolor same material prints.
2. The old way. It works well enough for many people, reduce tool change count.
3. Old way but fill any missing layer with primary extruder. This will keep tool change count low, at least as stable as 2nd way, does not fix material compatibility problem but should almost always work outside of very extreme combination.
4. Full separate ring for each extruder, adds unnecessary tool change to ensure tower continuity and stability.

Worth mentioning that printing internals of the tower with linear pattern is better since it is less likely to jumble together and create mess.

Let's ignore cases like 3 or more extruders where 2 are loaded with same material types (for example: PLA, PLA, PETG, PETG)

tommylux, 2021-11-21:
You could have a ‘ignore sparse layers’ if the wipe tower was at the back of the bed then the z axis can drop down to purge only when it needs to then return to full height to continue printing. This would use the print head size to avoid collision.

Ghostkeeper, 2021-11-24:
The original idea was basically to print a purge bucket and purge in there. It's indeed floating, but that was, according to our tests, not an issue. It falls into the bucket, and then gets wiped off as the nozzle moves back to the print.

> > Most common use is separable support.
> 
> Do you have a source for that claim? Because I very much doubt it's the single most common use.

Cura sends anonymous statistics about this (if the preference is not disabled). From that we can see that the most common use of multi-extruder printers is PLA+PLA, most likely multi-colour (we only see the material type, not colour). The combination PLA+PVA and PLA+ABS are also fairly common though (though you'd expect ABS to warp way too much for that, but oh well). And to my boss, it's worth mentioning that PLA+PVA is more common for Ultimaker printers.

We've discussed this with the team before and priming both every layer is an option for us (Block137's 4th option, but with the order changed to prevent an extruder switch). I also think that option 3 should get considered.

However all of these don't really get priority in the planning because the current solution works well as far as we can test. I'm convinced it will be different for other printers, but you'll understand that there is hardly any pressure on it from Ultimaker if there is no issue with their printers.

wawanbreton, 2021-11-24:
> The original idea was basically to print a purge bucket and purge in there

Ok, I actually didn't get it, now it looks more logical.

> However all of these don't really get priority in the planning because the current solution works well as far as we can test.

I am actually willing to work on this, even if the solution is very complex. Right now, I'm trying to find the solution that would best fit Ultimaker, Ultimaker users, and other users (including me :smiley:). My objective is to have the most minimal set of parameters that would fit every situation, so that we can do everything and it remains simple enough. For now, it would help me if many people gave a priority to each of these (1-useless => 5-mandatory) :
- Reduce tool changes as much as possible
- Reduce used filament as much as possible
- Do not print over nothing
- Do not print over a different material
- ... Any other feature you would consider very important
@Ghostkeeper It would also help to have your feedback, and maybe from you coworkers too.

On the other hand, I'm still thinking about the multi-materials problem : I only have PLA at home, so I can't test, but do you think the tower would remain consistent if we printed circles "inside" each other ? Let me explain :
- Set tool for E0 => print circles 8, 6, 4, 2
- Set tool for E1 => print circles 9, 7, 3, 1
This way, the secondly printed circles would be constrained inside the grooves created by the first one, so  maybe it would help them stick to each other. I really don't known whether it's a good idea...

Block137, 2021-11-25:
> My objective is to have the most minimal set of parameters that would fit every situation, 

IMO that will be impossible, or at least not optimal.
My printer is a tool changer so my priority order is like:
> 
> * Reduce tool changes as much as possible 
5 (Mandatory - each tool change is a mechanical wear and heat cycle)
> * Reduce used filament as much as possible
4 (pefered but not mandatory - one of the advantages of tool changer is no need for large purge block)
> * Do not print over nothing
3 (rarely a problem but when it is, is a pain - when PVA is wet for example, it expands violently and just won't go down the bucket)
> * Do not print over a different material
2 (only required for the main structure - if at least 1 ring is continuous single material, the tower will be stable enough for other rings to be assorted)
> * ... Any other feature you would consider very important
Bridging - use bridging if possible so the lower layer doesn't need to be solid but infill-like. (imagine a well-played Jenga)

>   This way, the secondly printed circles would be constrained inside the grooves created by the first one, so  maybe it would help them stick to each other. I really don't known whether it's a good idea...

It can't be worse. In my experience, single wall tower is plenty strong (even with 0.25mm nozzle) splitting them apart won't break it. The tower always release from bed before breaking (if you can improve that too)

Ghostkeeper, 2021-11-25:
I agree with Wawanbreton that it's useful to have a minimal set of parameters that fit every situation. Having too many parameters for tiny details is more likely to confuse users than to help them.

> * Reduce tool changes as much as possible -> 5: Takes long, interrupts flow (more purging) and leads to stringing
> * Reduce used filament as much as possible -> 2: Nice bonus, but the purged material is almost nothing compared to the print itself.
> * Do not print over nothing -> 4: While not a problem for UM, this is what most people seem to want here, possibly to prevent dragging filament along.
> * Do not print over a different material -> 1: As long as the outer shell of the tower is all in 1 material, the adhesion doesn't really matter.

Another property that I'd find important is to extend only, at least within reason. We prefer to keep the behaviour the same unless settings are changed, so it should be possible to keep the old behaviour and it should be the default too, unless it is more or less unanimous that the new behaviour is better.

> but do you think the tower would remain consistent if we printed circles "inside" each other ?

Since you want to prevent extruder switching for each of these circles, this also means that you'll print lines spaced apart first, then filling in the gaps with the other extruder. For PLA as well as other materials, I think you'll find that the lines spread out more with the first extruder than you intended. And as a result, there won't be enough room for the second extruder to prime. You might get more back-pressure when priming, so maybe it wouldn't prime as well.

wawanbreton, 2021-11-26:
> it should be possible to keep the old behaviour and it should be the default too

Understood, and agreed !

Thank you very much for the feedback ! I'm starting to see that there are 3 very different cases, depending on the printer :
1. 1 head, many nozzles, like Ultimakers : required priming is very low, no tool change required at all => in this case, changing the tool very frequently is not a problem, and whatever we do, the consumed filament will remain low.
2. 1 head, many tools, like Original Prusa XL : required priming is very low, but changing tool is very long => in this case, we should really avoid changing the tool, but the consumed filament will remain low.
3. 1 head, one nozzle, many inputs, like TiZYX with multi-materials head (mine)  : required priming can be VERY long, but changing tool is quite fast => in this case, what matters most is not consuming too much filament, because in some cases you can use more for the prime tower than for the model itself.
4. Other tricky printers ?

So indeed, the needs are very different. I think we can consider that cases 1 and 2 are similar, except that case 1 will always be faster. But case 3 is very different, because we really want to save filament when possible. So I think the main setting should be something like "Avoid changing tool as much as possible" :
- When enabled, prime tower is printed with current tool until a tool change is really required. Different materials may be printed on top of each other.
- When disabled, all the rings are consistent in material, and a tool change is done very frequently (should match the actual behavior)

And then, in order to save filament, an other option should be "Sparse layers infill" with a 0-100% value :
- 0% => spare layers are ignored, as currently
- 1-99% => spare layers are filled with an infill pattern (should it be customizable ?)
- 100% => spare layers are completely filled, making a fully dense tower

Tell me what you think, I'm writing as I'm thinking !

Block137, 2021-11-26:
My idea is a drop-down list with the following options
1. None (replace "Enable Prime Tower" checkbox)
2. Default (old behavior)
3. Same material (Prusa style, does not consider material incompatibility)
More can be added later
Then a drop-down named "Fill empty layer with" which you can choose an extruder, "None" or "Not-overridden" option (use whatever that reduce extruder change)

wawanbreton, 2021-12-05:
Hi, sorry for not answering this week, but I've been very busy at work.

> To get the most benefit out of it, I think it needs different tower than a tube.

Well, in the meantime I have spent some time to think about an optimal shape to fit the most important use cases. My conclusion is that it is very difficult to achieve with a round tower, especially if you want to create a hollow tower and use briding when you need to actually change tools. In this case, the Prusa model seems to be the most optimal. However, I don't like to copy existing things, so I'm still trying to think of something that would be as much efficient, but different. Maybe I will take some time and draw diagrams of my various ideas, so that I can have your feedback, but it is very time-consuming :)

Ghostkeeper, 2021-12-06:
Note that Cura used to have a square prime tower: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/dual/prime_tower_circular.md It was removed because the round one was considered to be better in every case, needing no sharp corners so the flow rate can be kept more constant. I don't really see a reason why a square one would allow greater freedoms than a round one and what it really has to do with skipping layers.

wawanbreton, 2021-12-10:
That's very interesting ! I must admit that a round tower has some advantages. It just does not help to create bridging, because distances are not regular all over the shape. So, the best idea I could come up with is a "wheel" infill, something like that :

[image]

That would keep the used filament very low, and provide a very coherent and strong tower (I think !). I would start filling the rings by the center, so that the first one would be layed on the existing center, and when rotating, each new ring would stick to the previous one, and be supported by some spokes of the wheel. However, I don't know if it would work well with a very large tower. The number of spokes shall be a setting. If not enough for large towers (with many different colors for example), we could split the spokes in different rings and have more spokes for the larger rings. If you find this solution interesting, I will take some time to make drawings of each use case.

tommylux, 2021-12-10:
Reminds me a little bit of a spiderweb. Nice original design idea.

[path846]<https://user-images.githubusercontent.com/601114/145630200-56284f80-58fc-41dd-b224-341f8a3d0c71.png>

wawanbreton, 2021-12-11:
> Reminds me a little bit of a spiderweb

Well, I didn't think about it, but that's very true, especially if you sub-split the spokes :
[image]

Actually, that would make sense, because the purpose of a spider is to maximize the surface of capture with no large holes, while using as few silk as possible. I should take some time and learn a bit about how they build their web.

tommylux, 2021-12-13:
Maybe a spiderweb design purge bucket maybe simpler to implement. I think the spokes should be consistent as this becomes the platform in which you can bridge material. A given colour may require more material to purge, in which case bridging is performed. For colours requiring less purge, only the spoke is printed conserving filament.

[path846]<https://user-images.githubusercontent.com/601114/145673618-fed35148-1036-4289-94a1-699b4a0032b5.png>

Ghostkeeper, 2021-12-17:
The problem with a wheel like that is that it'll need travel moves, which interrupt the flow too. That makes the prime tower less reliable to create a good flow at the start. A square tower would probably be better, because the flow interruptions from accelerations are probably shorter than the ones created with these travel moves.

I do see the point a square tower to improve bridging, thanks for the explanation! But I still don't think that the inside of the tower really needs to be sturdy.

wawanbreton, 2021-12-17:
> It's really nice to hear all of these ideas though :)

My pleasure !

> A square tower would probably be better

I was about to make a full diagram of the wheel infill, so maybe I will wait a bit :)

The wheel creates a very splitted workflow indeed, and that's really not good for clean printing. What I can think about to improve this is something like this :
[image]

With this rounded-zizgag design, all the "rings" are continuous and can provide a proper support. I don't know if it uses more or less filament than the previous one, but it is definitely more coherent and faster to print. Tell me what you think, but I don't think we can do much better with a round basis.

tommylux, 2021-12-17:
This design is suitable for 3 extruders like the A20T? Seems a little over complex that could be achieved using a square one with bridging, though a traditional spiderweb might be worth a look.

[path847]<https://user-images.githubusercontent.com/601114/146599133-2b9993f3-fb8e-4d37-9d98-ff9acb223a15.png>

wawanbreton, 2021-12-17:
Implementation seems very easy for every situation if that's what you mean. The square would probably be more efficient, but it actually looks "too easy" :laughing: 

> This design is suitable for 3 extruders like the A20T

I think so, these diagrams only represent the spare layers infill which would be printed with a single extruder, so no tool change. They do not have a link with the actual purge layers : the infill rings are splitted regularly, but you may have on top of it, a very large ring for a color and a very small one for the other. The purpose of the infill is just to make a support.

wawanbreton, 2022-01-09:
Hi again, and happy new year !
I finally took some time to draw examples of the various cases, so that we can discuss it very deeply. As there are many files, you can see/download them on this link : https://www.famille-mathieu.net/d/s/mzeofJbpWo2EJzIw8VfX6yND1QMXbWOi/SZY-L1yVqlbRnQ2boRoHSi-xpUHLtS-m-oL3gk_ErNQk with the password "wawanbreton"
What you can find in this folder : 
* The model.png is a very simple 3d view of the example file to be printed
* Each example sub-folder contains :
** A readme.txt to briefly explain the use case and result
** An output directory with the detail of every layer. I tried to make them as straightforward as possible, but don't hesitate to ask questions :)
The examples are very simple, but I think they are representative enough for you to extrapolate them to more complex cases, or even with 3+ extruders.

A few remarks/questions :
* I completely removed the inner and outer walls of the wheel infill. It actually looks like zigzag infill for supports, and it works very well even on long distances, so I assume this would work too. But an external opinion would be welcome !
* I'm not sure example3 would work very well with a Y-extruder, but I can't do much better, because in this case there is a time when I have both materials inside the nozzle !
* example1 mimics current behavior, but it still could be optimized a bit. But maybe it's irrelevant considering that it is used for multi-extruder case when optimization is not expected !
* example2 would be perfect for me :D

wawanbreton, 2022-02-09:
Well, any feedback on this ? Can you access the files at all ?

Ghostkeeper, 2022-02-26:
I could originally, although the link seems to have expired by now. It says (in French) that the files were configured to expire at a certain date. These files seem to have gotten lost.

From what I remember, my main thought was that I was wondering how any of these options would prevent layers from overhanging if some of the layers are skipped. I'd think that printing along the wall is much better, since it gives the line something to stick to and still involves fewer accelerations. But I didn't react because I wanted to see if other people had some feedback too, because perhaps I had been missing the point... There's a lot of information, after all.

wawanbreton, 2022-03-26:
Hi,

Sorry for the link, I give some more time, so you can access it again

> I'd think that printing along the wall is much better, since it gives the line something to stick to and still involves fewer accelerations

Probably, but that implies printing in mid-air in some conditions.

I think at this point, I will try an implementation, probably with hard-coded parameters, so that I can make a few preliminary tests and show what it makes in real conditions.

wawanbreton, 2022-05-14:
Hi, it has been some long time, so I wanted to publish some progress. I have done an implementation of the wheels pattern in the engine, which was quite fast, then it took me a looooong time to actually run the Cura GUI using this engine. Finally, this is an idea of my current status :
[image]

I still have a few bugs, but this is basically working. I will soon be able to start making real tests.

tommylux, 2022-05-14:
Noice!

[Screenshot_20220514_081816]<https://user-images.githubusercontent.com/601114/168413620-347ce046-2598-425c-84f7-58f8b983898d.png>

wawanbreton, 2022-06-09:
Hi,
I have been able to do some real-life tests ! Printed two things yesterday : the first one is a small gift box, which is an easy case because the 2 colors a required for each layer, except the end. So the prime tower is quite clean :
[image]

[image]

And then I wanted to test the toughness of the tower when no extruder change is made during a lot of layers. I made a useless test model, which performs quite well too :
[image]
[image]
[image]
However, my extruder had some trouble during a filament change at the end, and the grey could not be re-inserted. This explains why the top is not very clean, it misses some layers. Anyway, despite its height, the tower is still very strong and could have been even higher without any trouble.

So far, from what I can tell, I am very satisfied with the result. The second model was very fast too print, as there were very few extruder changes. And the towers are very light, which means a lot of filament has been saved.

Are there any other specific tests you would like me to do, to check something important ?

wawanbreton, 2022-06-11:
> Does the purge amount reduce when there isn’t a colour change? Ie structure only?

Yes indeed, when there is no need to purge, it only prints the wheel instead of full discs, which is much faster and uses less filament. You can (hardly) see it on the 3rd picture I posted.

> The tower would need to be positioned at the far end of the bed to prevent collision

I think the main philosophy of cura engine is never to go back down. Wherever you place the tower, there is a risk of collision, which completely depends on the shape of your printer. So there is no way to ensure no collision for every printer, unless Cura internally has a full 3D model of your printer and computes 3D collision, but that would be very CPU-time-consuming and hard to maintain ! And you can't let that to the user, because such a collision may damage your printer...

tommylux, 2022-06-11:
Sure. They do have this feature in PrusaSlicer, but you have to accept the risks. Have you made a pull yet for your build?

wawanbreton, 2022-06-11:
> They do have this feature in PrusaSlicer, but you have to accept the risks
Maybe this does work only with official Prusa printers, I'd be curious to know.

> Have you made a pull yet for your build
Not yet, I still have minor bugs to fix and I'd like to have feedback from @Ghostkeeper. It is only in preliminary testing step ! If you want to test it yourself you will have to build Cura and CuraEngine from my repositories.

tommylux, 2022-06-11:
It works with non Prusa printers. a20t etc.

[14 more replies]

closed 2026-09-10 by GregValiant