Ultimaker/Cura #6911: [4.4.1] Object floating are still sliced on the build plate

timherrm, 2020-01-03:
**Application version**
4.4.0, 4.4.1

**Platform**
Win 10 1903 Pro x64

**Printer**
Ender 3

**Reproduction steps**
1. Dropping an Object onto the buildplate.
2. Setting x to 10
3. Hit "Slice"
4. In Preview you can see the object is still sliced on the build plate

**Screenshot(s)**
[image]

**Actual results**
Print is sliced onto the build plate

**Expected results**
Print should be sliced "mid air"

**Additional information**
"Automatically drop models to the build plate" is disabled under settings.
I don't know which version i had before, but there it worked fine.
I don't want to print in mid air, but onto another 3d print, that is why i need the z space.

Thanks!

rburema, 2020-01-07:
Confirmed, in latest master as well.

fieldOfView, 2020-01-07:
See the "Remove Empty First Layers" setting ("Mesh Fixes" category)

rburema, 2020-01-08:
Yes, that does the trick. So, not a bug but expected behaviour.

closed 2020-01-08 by rburema