Ultimaker/Cura #6912: Anycubic 4max pro - printer settings.

aux-lux, 2020-01-03:
**Is your feature request related to a problem? Please describe.**
"Anycubic 4max pro" printer settings are missing, and I couldn't find proper settings online. I have issues with end gcode, after print it smacks on a wall without end-switch.

**Describe the solution you'd like**
Anycubic 4max pro - printer settings available in Cura.

**Describe alternatives you've considered**
Provide printer settings, that could be entered manually.

**Affected users and/or printers**
Only Anycubic 4max pro

Have a good day kind people :)

fieldOfView, 2020-01-06:
The problem is, we don't have those settings either. We also don't have the printer, so even if we came up with settings we could not test them.

Ultimaker employees generally only have access to Ultimaker printers, so they only make profiles for Ultimaker printers. (Almost) all other printer profiles have been provided by community members (with access to those printers) or the manufacturers of the printers. You should really be asking Anycubic, or a community of Anycubic 4max pro users.

> Have a good day kind people :)

Thanks!

rburema, 2020-01-07:
Like @fieldOfView says, there's not much we can do unless somebody makes a pull-request.

I can't promise I can help, but what is your current end gcode?

aux-lux, 2020-01-07:
I modified it a bit and currently its:

> M104 S0 ; turn off extruder
> M140 S0 ; turn off bed
> M84 ; disable motors
> M107
> G91 ;relative positioning
> G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
> G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more
> G1 Y180 F2000
> M84 ;steppers off
> G90

It's not perfect, but works, kinda. By default I use most of gcode from Anycubic 4Max (not Pro). I removed some movement, not sure which, and removed beeping. But maybe you could help me to make it a bit more "useful". Stage could go few steps down in Z axis (if not already at the bottom), and home in X axis (so it will be protected with end sensor). That would be decent for me :) Yet I'm not sure about the rest of the printer settings. I'm not sure how much those printers differ. I found some profiles online to that printer (anycubic support team seem to be the source).

closed 2020-01-09 by nallath