Ultimaker/Cura #6913: [4.4.1] Nozel 2 support material missing from floating part

jibb348, 2020-01-03:
**Application version**
4.4.1

**Platform**
Windows 10, NVIDIA Quadro K3100M

**Printer**
Ultimaker 3

**Reproduction steps**
1. Built simple box with rounded edges in Solidworks
2. Exported as STL
3. Imported into Cura
4. Rotated to place bottom face on print bed
5. Wanted to avoid the glossy build plate finish on the underside so turned off 'automatically drop models to build plate'
6. Lifted model ~2mm
7. Sliced the model with Black PLA in extruder 1 and PVA in extruder 2
8. Examined the support material in the preview 
9. A ring of support material appears to be missing from the outer edge of the part between the build plate and the part.

See below for a view from beneith the sliced model
[image]

**Expected results**
I would expect the support material to completely support the base of the box

**Project file**
[UM3_TestBox.zip](https://github.com/Ultimaker/Cura/files/4020934/UM3_TestBox.zip)

Here's the source STL file as well:
[TestBox.zip](https://github.com/Ultimaker/Cura/files/4020937/TestBox.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I appreciate that this is a bit of an odd thing to do (just to improve the bottom surface of the model but would have expected the support material to cover the complete area.

This problem seems to go away if you lift the model higher off the print bed. Any reason for thin support areas to be culled?

Many thanks for looking at this.

nallath, 2020-01-06:
I've seen this happen a few times before, it seems to have something to do with the "stair stepping" algorithm. It's on our backlog to have a look at that.

Alternatively, you could just also set your platform adhesion to raft, which essentially does what you want, all the while providing you with more control (because then you can modify the raft settings seperately from the support settings)

rburema, 2020-01-07:
Confirmed as yet another stair-step duplicated/related problem :-(

Current workaround for such cases is to set low to 0 values for either 'support stair step height' and/or 'support stair step maximum width'.

jibb348, 2020-01-08:
Hi Nallath, Rburema,

Thanks for the help. Will use raft from now on for this situation.

Ghostkeeper, 2020-01-13:
Duplicate of https://github.com/Ultimaker/Cura/issues/6379.

closed 2020-01-13 by Ghostkeeper

rburema, 2020-06-25:
We might finally have squashed this bug! When we eventually release 4.7 (which might take a bit longer than usual) you can check it out.