Ultimaker/Cura #6914: Plugin Inventor Dosn't Work

Beppo86, 2020-01-03:
the plugin that should read the inventor files, from version 4.4.1 is no longer present on the store.
serves in remedy or a procedure to reactivate it
thank you

fieldOfView, 2020-01-03:
That plugin is a 3rd party plugin by @thopiekar, and it needs to be updated to be compatible with Cura 4.4 and newer.

Beppo86, 2020-01-03:
thanks, but I didn't think it was an external plugin, since ultimaker advertises it very evidently

fieldOfView, 2020-01-03:
It is though:
https://thopiekar.eu/cura/cad/inventor/

Beppo86, 2020-01-03:
do you know if there is a method to add this extension? because there is nothing on that page.
or do you know if you can download the old version? and if so, can you also give me the link?
thank you

fieldOfView, 2020-01-04:
The old version does not work with Cura 4.4

You can download Cura 4.3, and use the Inventor plugin in that version of Cura.

rburema, 2020-01-07:
If you want to contact @thopiekar regarding this plugin, please do so here: https://github.com/thopiekar/CuraInventorPlugin/issues

closed 2020-01-07 by rburema