Ultimaker/Cura #6915: mixed layer height prints give bad layer height results

nathanielstenzel, 2020-01-03:
With Cura 4.3.0 or 4.4.0 or 4.4.1 and probably older, I get odd results in the Z positions for the gcode.

Layer Height of 0.14
Infill Layer Thickness of 0.21

This results in 0.28 differences in Z. The parts have been coming out brittle and with wispy infill due to this.

I see a number of other Z position related bug reports. They may all be connected.

nathanielstenzel, 2020-01-04:
0.21 layer heights for walls, infill and support results in 0.35 increments in Z.

nathanielstenzel, 2020-01-04:
;Generated with Cura_SteamEngine 4.4.1
M666 X0 Y0.2 Z0.3 ; Add your endstop values here after calibration
G29 P2 Z0.28 ; Auto-level ; ADJUST Z higher or lower to set first layer height. Start with 0.02 adjustments.
G1 Z30 ; raise Z 30mm to prepare for priming the nozzle
G1 F3600 Z2.21
G0 X-10.445 Y4.539 Z2.21
;TYPE:SKIRT
G1 F3600 Z0.21
;TYPE:WALL-OUTER
;TYPE:WALL-INNER
;TYPE:SKIN
G0 F3600 X5.242 Y-6.071 Z0.48
;TYPE:WALL-OUTER
;TYPE:WALL-INNER
;TYPE:SKIN
G0 F3600 X5.261 Y6.052 Z0.83
;TYPE:WALL-OUTER
;TYPE:WALL-INNER
;TYPE:SKIN
G0 F3600 X5.245 Y-6.068 Z1.18

nathanielstenzel, 2020-01-04:
;Generated with Cura_SteamEngine 3.6.0
G1 Z15.0 F6000 ;Move the platform down 15mm
G0 F3600 X12.455 Y12.677 Z0.3
;TYPE:SKIRT
;MESH:cylinder.stl
;TYPE:WALL-OUTER
;TYPE:WALL-INNER
;TYPE:SKIN
;MESH:NONMESH
;MESH:cylinder.stl
G0 F5400 X2.905 Y-8.937 Z0.45
;TYPE:WALL-INNER
;TYPE:WALL-OUTER
;TYPE:SKIN
;MESH:NONMESH
;MESH:cylinder.stl
G0 F7200 X2.905 Y-8.937 Z0.6
;TYPE:WALL-INNER
;TYPE:WALL-OUTER
;TYPE:SKIN
;MESH:NONMESH
;MESH:cylinder.stl
G0 X2.905 Y-8.937 Z0.75

fieldOfView, 2020-01-04:
> Layer Height of 0.14
Infill Layer Thickness of 0.21
>
>This results in 0.28 differences in Z. 

This is by design. This is the description of the Infill Layer Thickness: "The thickness per layer of infill material. This value should always be a multiple of the layer height _and is otherwise rounded_." (my emphasize). So in your case, the Infill Layer Thickness is rounded to 0.28, the nearest multiple of 0.14. 

Your snippets of gcode are abbreviated too much for me to see what is going on.

nathanielstenzel, 2020-01-04:
You say that in the one example, the infill should be 0.28 height changes but it is making ALL height differences 0.28.

fieldOfView, 2020-01-04:
What should be happening is that basically the infill should skip each other layer. So there should be a layer with skins and infill, and then a layer with just skins. Then a layer with skins and infill, followed by a layer with just skins. Etc.

If you could post a complete gcode file, we can see what is going on.

nathanielstenzel, 2020-01-06:
It seems I did indeed have a misunderstanding about how Cura works for this feature. I think Slic3r did it better and was able to do 0.2mm walls with 0.3mm infill and it just did walls or infill twice sometimes to get the needed results. By your description, Cura can only do the walls multiple times and can not do infill multiple times.

The other issue of 0.14mm walls and 0.21mm infill resulting in all 0.35mm Z differences was a result on a Cura profile that I switched away from and sadly I no longer have those particular settings or the file. Since I can no longer provide any evidence of that, I guess there is no point in talking about it further.

For now, I will settle for chunky 0.28mm layer heights for infill and support and 0.14mm layer heights for walls, but I hope that in the future you will let me do 0.21mm layer heights for walls with 0.28mm or 0.35mm layer heights for infill and support. 0.42mm layer height with a 0.4mm nozzle would only be doable with an increased flow or an increased extrusion width, both of which would be an increase in extruded filament and would be an experimental way of handling things.

Ghostkeeper, 2020-01-13:
Ah, I see that you opened a new issue here: https://github.com/Ultimaker/Cura/issues/6927

Let's continue the discussion there then.

closed 2020-01-13 by Ghostkeeper