Ultimaker/Cura #6916: Retraction Minimum Travel is lost in Cura 4

RealMook3d, 2020-01-04:
Hello!
**Is your feature request related to a problem? Please describe.**
I'm frustrated that Retraction Minimum Travel was removed from options.

**Describe the solution you'd like**
It would be good to take it from Cura 3.3.6 (the latest 3* stable, right?) and make it open to users in Cura 4 and further versions. Place it in the "Expert" level, please. So it will change visually nothing for many many users.

**Describe alternatives you've considered**
I understand that probably it was removed to make the UI less cluttered with different options but the behaviour that now Cura decides where to retract or not is wrong. I've found cases where it "draws" just a circular shape (between other circles) and makes breaks with slightly angled straight lines which are produced with retracts. It's just an example. Some other excessive unwanted retract usages are highly possible. This wears off a filament, makes clogs more possible and stresses an extruder more when it's not necessary at all.
The default 0.8mm in 3.3.6 is way too small. For technical parts such a "jewelry" setting is unwanted in many cases. Cura already has different settings accessability levels. So if somebody is "scared" with a lot of new terms and options, it's always a choice to use "Basic" or "Advanced". 

**Affected users and/or printers**
"Makers" will benefit definitelly! I love Cura 4 for "Small Feature/Hole" option a lot. But it's missing that Retraction "fine tune" min. travel option. I've noticed that it was missing starting from some version. I was just searching and was sure that it was somewhere... But it's just gone. Feeling frustrated.

**Additional context**
Another user noticed that Cura 3.3.6 has a better gcode generation for bridges. I'm not sure what's the difference but my guess that it (maybe) has some troubles. Will be checking in few days. I see sagging in round windows while using this model https://www.thingiverse.com/thing:2893943
That wasn't the case in 3.3.6. 
I know that if "strictly" then it's not in the context. But it's another reason why I possible switch back. And it's sad.

JustCost, 2020-01-05:
Support. A huge request to the developers of "Cura" to return this function and make it available for any printer. I can't turn it on. Even if the printer settings are custom.

rburema, 2020-01-07:
It's still there, it just isn't visible. (You can go into 'manage settings visibility' and see that it's probably hidden by the currently active machine, since it's not _explicitly_ in one of the categories in the visibility profiles.)

Should be in the 'travel' category next release. Fixed in commit: https://github.com/Ultimaker/Cura/commit/9c0bd007144cf3249e2ff587c693820d1f6d6e2a

closed 2020-01-07 by rburema

RealMook3d, 2020-01-07:
I've created Custom FFF printer. A lot of settings has (i) sign and "hidden by the currently active machine". The same as I had with the previous printer activated.
What's the criteria for this? Still can't get this to work. Also a lot of handy features are unavailable to choose. Is it a bug?

JustCost, 2020-01-07:
I also have this problem. The settings are hidden by the current printer - although I created a custom printer (all settings should be open) I think this is a bug of developers.