Ultimaker/Cura #6919: [4.4.1] Recent files in Windows Jump list cannot locate files correctly

imonsei, 2020-01-05:
**Application version**
4.4.1

**Platform**
Windows 10 

**Printer**
Irrelevant 

**Reproduction steps**
1. Start cura
2. Open a stl file
3. Close cura
4. Right click cura icon, either in start menu, or on taskbar
5. Click one of the recent file entries

**Screenshot(s)**
[image]

**Actual results**
Error message pops up asking user if file perhaps was moved

**Expected results**
Cura opens, loading file clicked in recent list.

**Project file**
Irrelevant 

**Log file**
Did not find anything relevant in log file.

**Additional information**
I did make sure I hadn't changed the files or moved/deleted them.

rburema, 2020-01-10:
I can confirm this happens in 4.4, and that 4.3 works correctly.

This is going to be a difficult one, since this happens _before_ Cura starts _and_ the right file is clearly known to windows.Yet it's still probably somehow related to something we did, since 4.3 still works.

I've found where windows saves this information, but those file don't seem to do anything different between the versions at first glance at least.

Ghostkeeper, 2020-01-13:
Maybe something to do with quotes and spaces in the file path?

ninovanhooff, 2020-01-13:
Developers: https://jira.ultimaker.com/browse/CURA-7102

goscicki, 2020-05-15:
I have been having this exact problem ever since I uninstalled all old versions of Cura prior to installing 4.5. I still have this problem with 4.6.1.

rburema, 2020-05-15:
It's still on the bug backlog I'm afraid :-/

We'll pick it up at some point, though that might be a while (we've got a lot on our plates).

goscicki, 2020-05-15:
Ok. I just thought it might be helpful to know that it stared when I removed old versions of Cura.

rburema, 2020-05-15:
During the installation or afterwards?

goscicki, 2020-05-15:
I removed old versions before installation. Then noticed the problem after installing the new version (4.5 at the time).

goscicki, 2020-05-15:
An interesting clue I just noticed by having all versions installed side by side...
[image]
The file icons are missing on 4.4+, but also the program icon is missing on 4.4.

evtrados, 2020-10-09:
Hi @goscicki,

I think I've found the reason behind your issue. Since, indeed we had not changed anything in that part of the code, I did a little test of my own because the earliest version I had installed was Cura v4.3.0.
I first opened the file explorer and clicked on 'Quick Access', so that my recent files would be displayed at the bottom of the window. If you do that and select one of the .stl files it should either:
- Open with Cura v4.3.0, or 
- Display the pop up window that asks you with which program you want to open this file with. There whatever Cura you choose it will still open it with 4.3.0.

That happens because in the registry the stl file opening is associated with all Cura versions you've installed, but it will open only with the earliest one (in your case 4.3.0). So, if you want to remove this registry associations completely you should open regedit as an administrator and then remove the following:
- All the HKEY_LOCAL_MACHINE\SOFTWARE\Classes\UltimakerCura.stl* folders. Multiple will exist, one for each version you had installed
- All the HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.stl folders. Again there'll be one for each version you had installed on your system.

Now you'll need to right click on an .stl file and choose 'Open with' and then select the latest Cura you're using. After that's done, the recent files will be opening with the version you desire.
Let me know if that helps.

rburema, 2020-10-09:
@evtrados Great! This one really stimied me. Can you add it to the change control board? I think we might need to discuss if we can solve problems like this in a version upgrade then.

evtrados, 2020-10-09:
Yes, I will add it to the board. As far as I investigated these entries are not deleted upon uninstall. The only one that gets deleted is the "actual" 'Open with..' association in HKEY_CLASSES_ROOT\.stl\OpenWithProgids, so we might need to add the aforementioned ones as well.

goscicki, 2020-10-09:
@evtrados Thank you! I haven't installed 4.7 yet because I'm almost done with a big important project. If I'm done tonight, I'll install 4.7 and delete those keys. I'll be sure to let you know how it goes. Thanks again!

goscicki, 2020-10-10:
@evtrados Unfortunately the problem persists after following those steps. I should clarify. The problem doesn't seem to be with file associations, since Cura opens with clicking directly on a file. The problem is in the recent jump list on the windows task bar. 

[image]

evtrados, 2020-10-12:
Thanks for trying it out @goscicki and I'm sorry it still doesn't work for you.
Yes, I also focused on the recent jump list, but (in my case at least) I found out that these files were not found because they were associated with another (earlier) Cura version.
So just to check one thing, if you go to File Explorer --> Quick Access and double click on the TPI_v9.3mf file there, with which version of Cura does it open?

goscicki, 2020-10-12:
That opens with 4.7 as expected.

goscicki, 2020-10-12:
I should also note that I deleted the following keys and rebooted:

HKEY_LOCAL_MACHINE\SOFTWARE\Classes\UltimakerCura.stl* folders
HKEY_LOCAL_MACHINE\SOFTWARE\Classes\UltimakerCura.3mf* folders
HKEY_LOCAL_MACHINE\SOFTWARE\Classes.stl folders
HKEY_LOCAL_MACHINE\SOFTWARE\Classes.3mf folders
HKEY_CLASSES_ROOT.stl\OpenWithProgids
HKEY_CLASSES_ROOT.3mf\OpenWithProgids

goscicki, 2020-10-12:
Got it! It took one last step. I deleted TPI_v9.3mf from the jump list, then reopened the file. That re-added it to the list correctly and it works now. Thank you so much for your help!

evtrados, 2020-10-16:
Glad that I could help @goscicki!!
And great that you got it working! :)

goscicki, 2020-12-19:
Have the same problem after installing 4.8. 

I noticed that HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.3mf was still missing, so I added the following:

[HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.3mf]
@=".3mf.4.8"

[HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.3mf\OpenWithProgids]
".3mf.4.8"=""

[HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.3mf.4.8]
@=""

[HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.3mf.4.8\DefaultIcon]
@="C:\\Program Files\\Ultimaker Cura 4.8.0\\Cura.exe,0"

[HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.3mf.4.8\shell]
@="open"

[HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.3mf.4.8\shell\open]

[HKEY_LOCAL_MACHINE\SOFTWARE\Classes\.3mf.4.8\shell\open\command]
@="\"C:\\Program Files\\Ultimaker Cura 4.8.0\\Cura.exe\" \"%1\""

Now files open with 4.8 however the jump list for 4.8 still does not work. But 4.7 still works fine. I am so confused.

goscicki, 2020-12-20:
Well, I found a solution for now by making this change:

[HKEY_LOCAL_MACHINE\SOFTWARE\Classes.3mf]
@="3mf_auto_file"

Now the jump list works on 4.8 but not on 4.7, which is ok for now. This is a somewhat confusing issue that I have never experienced with any other program.

evtrados, 2020-12-22:
Hmmm, it is a bit confusing indeed. I'll try to investigate why it happens. It's probably because there's something going wrong with assigning the latest version as default for all the file types.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

imonsei, 2024-11-23:
> I'm cleaning house. Is this still an issue in current Cura versions (5.8.0 and up)? Can this be closed?

I do not know. I have stopped using Cura, so I am unable to assist with testing.

GregValiant, 2024-11-23:
I'll close it.

closed 2024-11-23 by GregValiant