Ultimaker/Cura #6920: Screen flicker at 144hz

kulcris, 2020-01-05:
**Application version**
(The version of the application this issue occurs with.)
4.4.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
windows 10, nvidia gforce rtx 2080 

**Printer**
(Which printer was selected in Cura?)
NA

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
Have 144hz monitor, run cura. when cura is focused all windows flash constantly (dark mode) i thought maybe my printer was causing some power fluctuations but it happens only when cura was up

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
screen flickers slightly almost as if the monitor is going out.

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
Monitor is only a couple of months old and does not have this issue with any other app

Mr-Newlove, 2020-01-08:
I am having the same issue, on my 3 monitor setup it. only ever effects the screen it is open on. and is way worse when you move the mouse around, resize the window or cause a redraw of any sort.

I am running Cura 4.1, only started happening recently. I think it started around the time the Nvidia drivers supporting adaptive sync/freesync came out but disabling the feature doesn't affect it. I have an RTX 2070 and an updated version of windows 10.

Mr-Newlove, 2020-01-12:
Just an update, a new graphics driver came out a few days ago for me. V441.87 actually fixed the issue. I no longer have flickering even at 144hz

fieldOfView, 2020-01-12:
Thanks for the update. Does this also fix the issue for you, @kulcris?

Ghostkeeper, 2020-01-13:
So far this seems like a driver bug to me.

We can't reproduce the issue at work because our work doesn't supply us with those fancy GPUs or monitors :stuck_out_tongue:

kulcris, 2020-01-14:
Seems issue has been resolved with latest driver update. Will update if this starts to happen again

closed 2020-01-14 by Ghostkeeper

dxgldotorg, 2020-09-07:
I'd like to point out it never happened to me until I bought my 4k LG monitor, then the screen now sometimes blinks when Cura comes into focus.  It's something to do with the NVIDIA OpenGL ICD and its interaction with the display driver, most likely nothing @Ghostkeeper can fix.

I personally run a GTX1080, soon to get an RTX3080 and will see if it persists then.

kulcris, 2020-09-07:
Happened on my 2080 so I'm sure it will persist.

dxgldotorg, 2020-09-07:
@kulcris I personally have similar issues sometimes when closing an application (including my own DXGL library) that uses OpenGL, however it does seem to happen more frequently if my monitor is in HDR output mode.

Ghostkeeper, 2020-09-09:
I'm using RTX2070, refreshing at 120Hz, and I'm not seeing it.

It was put down as a driver issue because of the report that a driver update fixed it (and indeed this is the sort of issue more often arising from a driver issue than a programming error).