Ultimaker/Cura #6921: Malyan M200: change in End G-Code

willie68, 2020-01-05:
With the current end of G code, the extruder moves to position X0 Y127. At position X0 the extruder is near the wall and the filament is bent with a small radius so that it is not easy to get it out of the guide tube.
X120 would be a better alternative.
With Y127 the bed moves forward and over the actual end. Y120 would be sufficient.
The end of the G code should therefore be better:
G0 X120 Y120; (Stick out the part)

rburema, 2020-01-07:
Fixed it in the commit you see above (we're wholly reliant on the community for fixes in 3rd party printers, so we just have to trust you here).

closed 2020-01-07 by rburema

willie68, 2020-01-07:
Just have of the way. 
as metioned before, X120 would be good, too.
G0 X120 Y120; (Stick out the part)

willie68, 2020-03-03:
Just installed the new version, perfect.
Thanks.