Ultimaker/Cura #6922: not loading cura metadata from 3mf

kcris, 2020-01-05:
**Application version**
4.4.0

**Platform**
windows 10, 64 bit

**Printer**
none

**Reproduction steps**
1. load 3mf file which contains metadata, attributes under the correct cura namespace `http://software.ultimaker.com/xml/cura/3mf/2015/10`
2. file is loaded but metadata is ignored because the namespace prefix is not literally set to `cura`
(which is incorrect, cura is an xml namespace prefix, it acts like a handle for the namespace, as long as the namespace is correct, the namespace prefix should make no difference whatsoever when loading the 3mf xml)

**Screenshot(s)**
I have used a 3mf file with this model contents
```
<model... xmlns:customXMLNS0="http://software.ultimaker.com/xml/cura/3mf/2015/10" ...>
  ... <metadata name="customXMLNS0:extruder_nr" type="string" preserve="1">0</metadata>
</model>
```

**Actual results**
Model will be loaded but cura metadata is ignored.

**Expected results**
Cura namespace (not the prefix) is correct = http://software.ultimaker.com/xml/cura/3mf/2015/10.
Metadata should be loaded independent of the namespace prefix used (in my case `customXMLNS0`).

However, by changing the namespace prefix to `cura`, Ultimaker will correctly load the metadata, which means that you are searching for a specific namespace prefix and not the prefix itself.
And your code confirms this: 
This is incorrect according to standard xml functionality. Loading should work the same, no matter what namespace prefix is used.

**Project file**

**Log file**

**Additional information**
see plugins
* plugins/3MFReader/ThreeMFReader.py
* plugins/3MFWriter/ThreeMFWriter.py

Ellecross, 2020-01-13:
Hey @kcris  we tried to change the cura namespace prefix to something else in the 3dmodel xml file, but the metadata seems to be loaded for us using Cura 4.4.0. 

Can you share a 3mf file with us where we can reproduce this? Thanks.

Ghostkeeper, 2020-01-13:
Maybe has something to do with our XML material profile loader looking for namespaces using `iterfind("cura:setting", __namespaces)`? In the Python documentation they are using `iterfind("{http://some/namespace/}tagname")`, but also saying that the second argument maps from namespace prefixes to full namespaces, so I'd expect the current solution to work according to spec.

kcris, 2020-01-13:
Hi,
attached two 3mf files which will show the problem
* first file was exported via lib3mf which is using a custom namespace prefix when exporting the metadata, this namespace prefix is `customXMLNS0`. 
* second file is a copy of the first file, where I patched the contents of the 3dmodel.model file inside the 3mf, to replace that namespace prefix `customXMLNS0` with `cura`, nothing else was changed

These files contain some metadata values, e.g. `infill_sparse_density` is set to a value of 60.

Now, load first file into cura 4.4.0
- chose any mesh
- go to "Per Model Settings" / "Modify settings for overlaps" / "Select settings" / "Infill" - check "Infill Density"
- you will see that Cura shows a value of 20 for the infill density, which is not 60 (I guess 20 is some cura default value), which means it has ignored my metadata values

Close model, do the same steps for the second file
- you will see that Cura shows a value of 60 for the infill density, which is the value I set, which means it has correctly loaded my metadata as soon as I used the "cura" namespace prefix
[issue6922.zip](https://github.com/Ultimaker/Cura/files/4056333/issue6922.zip)

Ghostkeeper, 2020-01-14:
Uh, right. I didn't read the original issue well.

Cura doesn't read slice settings from the metadata tags of objects in the 3MF file. In fact, Cura doesn't look at the metadata at all. Instead, Cura Projects are an extension on 3MF that puts the profiles (entire files) separately in the archive. It reads those.

As an example, take this file: https://github.com/Ghostkeeper/SlicerTestModels/blob/master/cura_projects/4.4_custom_fff.3mf

If you open that with an archiver application you'll see that it contains an additional folder called "Cura" that contains a bunch of profiles. These are the profiles that Cura will actually read.

kcris, 2020-01-14:
> Cura doesn't read slice settings from the metadata tags of objects in the 3MF file. In fact, Cura doesn't look at the metadata at all. Instead, Cura Projects are an extension on 3MF that puts the profiles (entire files) separately in the archive. It reads those.
Hi, I am not following you I guess,

I sent two files in my previous comment, 99% identical, both including metadata, both being standard .3mf files 

First is loaded into cura and metadata ignored.
However, second file is fully loaded into cura!!

The only difference: the namespace prefix.
- first file uses a custom namespace prefix `whatever` for the metadata
- in the second one I replaced that custom namespace prefix with `cura` (which shouldn't be required according to xml general rules)