Ultimaker/Cura #6923: NMAKE : fatal error U1073: don't know how to make 'install'

ArvinShaffer, 2020-01-05:
Running Cura from Source on Windows with a problem at Cura-build CMake workflow step 6

**Application version**
4.4.1

**Platform**
Microsoft Visual Studio 2015 
windows7 sp1

**Printer**
no Printer

**Reproduction steps**
1. i just do the step in wiki, Running Cura from Source on Windows
2. and i do the step in Cura-build CMake workflow
3. when i do the step till step 6 Run nmake install to install the dependencies.
4. i got a problem ,NMAKE : fatal error U1073: don't know how to make 'install'

**Screenshot(s)**

[image]

**Actual results**
NMAKE : fatal error U1073: don't know how to make 'install'

**Expected results**
nmake install is ok

**Project file**

**Log file**

**Additional information**

nallath, 2020-01-06:
We build Cura with MINGW, not Visual Studio. So I'm not that suprised that NMake doesn't work. 

I'm not sure if there is anyone in the team that tried this (and also got this to work). Is there a specific reason that you want to use visual studio instead of mingw?

ArvinShaffer, 2020-01-07:
> We build Cura with MINGW, not Visual Studio. So I'm not that suprised that NMake doesn't work.
> 
> I'm not sure if there is anyone in the team that tried this (and also got this to work). Is there a specific reason that you want to use visual studio instead of mingw?

i just follow the step of Running Cura from Source on Windows i have install the mingw, and use to compile of it, so you think i could use make install instead of nmake install ?

LipuFei, 2020-01-07:
@ArvinShaffer The cura-build-environment repository has not `install` target. If you run make, it will compile and install all components/dependencies to `CMAKE_INSTALL_PREFIX`

Ghostkeeper, 2020-01-13:
This is expected behaviour right now. The reason for it is because the purpose of that repository is to build the dependencies, so we call the build scripts of the dependencies individually which already installs them. It needs to install them directly since they depend on each other (e.g. SIP vs. PyQt) so one package needs to be installed before the other one can even start building. This way we don't need to modify the actual build scripts of the dependencies, which would be a lot of work and very fragile.

closed 2020-01-13 by Ghostkeeper