Ultimaker/Cura #6925: No way to correlate Acceleration to motor RPM

Wdfitzpat007, 2020-01-05:
4.4.1

The problem with setting acceleration values globally with respect to RPM is that stepper motors to not work this way. As RPM increases stepper motor torque decreases exponentially.

It is my opinion that this is one of the main causes of the ringing you see when setting higher acceleration values. Furthermore, lets look at the other main causes: Insufficient machine rigidity, backlash, and unknown nozzle pressure. Also lets bring up another common print issue: lost steps. Lost steps occur when load surpasses motor capacity.

Doing this will lower the load placed on the steppers when at their weakest, but without limiting it when at maximum strength. Now lets see how setting a dynamic acceleration corresponding to RPM impacts each of the aforementioned issues.

-Insufficient machine rigidity 'slop'. Cheap Chinese POM rollers have more slop when you push on them and apply higher than usual load, right? If we lower the load on the motors when moving fast, a side effect will be lowering the load on the linear motion system. While I have seen no empirical studies demonstrating this, and therefore cannot unequivocally say that this will positively impact print quality when running fast. Lets not kid ourselves here either. Another consideration is the timing belt, when moving faster timing belts have a lower load capacity. I know this because I work engineering linear motion systems and have proven several of my designs inadequate after the fact because I failed to check the load rating at a given speed of the belt width and pitch I was intending to use at the length I was planning on using it at.

-Backlash. The reason you do not see widespread issued with backlash in threaded rods is because the downwards load never exceeds the load of gravity placed on the assembly. This is also why no-one EVER needs to use anti-backlash nuts, ballscrew or threaded rod. In fact depending on the kind of anti-backlash screw and the strength of the spring, you could decrease the load capacity of the system without incurring appreciable backlash. For example if you had one of the oil impregnated bronze anti-backlash nuts with the spring side facing down with gravity causing the spring to hold the mounting side to the top profile of the threaded rod against gravity. In that use case if the screw is able to exert less than twice the force of the gravity load without any appreciable deflection you will increase the backlash of the system. The spring end would have to facing 'up' pressed against gravity with the mounting side facing down the spring pressing with gravity. Ballscrew is only better than leadscrew if it has a higher accuracy over distance.

-Backlash part 2 belts. To address confusion on belt reinforce type, the specific modulus of Kevlar 149 is 127 GPa/ (gcm^3), S-glass (aka S-2) fiber 35GPa/ (gcm^3), 1045 steel 26GPa/ (gcm^3). Values used kevlar 186/1.47, glass 89/2.53, steel 205/7.87. This means that for the same level of stiffness a kevlar belt will have 10% higher cross sectional area but only weigh 21% of what the steel belt weighs. Thus making the steel belt 5 times heavier for no reason. Furthermore when using steel the strand size is far greater than kevlar resulting in the belts being far stiffer chewing them when going around small pulleys. Look up "Steel Core Polyurethane belt used with a 16 tooth pulley", there are pictures, real nasty. 

Deflection in the belt is not where the error will come from, the error comes from not using anti-backlash sprocket. GT 3mm sprocket for example have 114 micron worth of clearance. GT 2mm will have less clearance, but I do not have a solid number off hand. Friction, belt pitch, belt pre-load, resulting belt compression (squish into the tooth clearance), and tooth engagement (a function of belt pitch and width) have a role in how much the tooth clearance effects the overall slop of the machine. So it is not as simple as saying that you are limited to 114 micron of accuracy with GT 3mm. Something more to the effect of you have within half the tooth clearance of the drive in each direction. The less load placed on the belt, the less the teeth will wiggle in the drive grooves.

-Calmer decelerators will give the nozzle pressure more time to come down before stopping. Making the plastic flow more predictable.

-Lowering the load when the motors are weaker lowers the load. This is self explanatory it would take more force to cause a lost step without using a larger stepper or unnecessarily increasing the print time.

Making the acceleration dynamic would have many different advantages. Some in-substantial, others very meaningful.

fieldOfView, 2020-01-06:
Do you know that Acceleration and Jerk in fdm printing are defined differently that in physics? They are basically misnomers. You don't specify an acceleration, but a maximum instantaneous change in velocity. The printer firmware does not know about true acceleration.

Wdfitzpat007, 2020-01-06:
No, I am not familiar with the way acceleration or jerk settings are applied at a firmware level. Perhaps this suggestion is better suited for a firmware level change?

fieldOfView, 2020-01-06:
Probably. Either https://github.com/MarlinFirmware/Marlin or https://github.com/dc42/RepRapFirmware

The firmware that Ultimaker uses is based on an old version of Marlin. But features and fixes to Marlin are rarely (if ever) backported to the Ultimaker fork of Marlin.

closed 2020-01-06 by Wdfitzpat007