Ultimaker/Cura #6926: Crashing on startup,  both 4.4.0 and 4.4.1 (sucessfully reloaded 4.3.0)

dcharbonnet, 2020-01-06:
I have used Cura for most of a year and it has worked fine.  The new version crashed on start up.  I have tried this many times, redownloading the files and rebooting the computer.  Crashes every time.  Sent in 4 crash reports.

**Application version**
4.4.0 and 4.4.1

**Platform**
AMD Phenom II X4 955 processor

**Printer**
Anycubic Mega 3, unless it picked it up from the previous installation, I didn't enter this.

**Reproduction steps**
1) download install file
2) run install file
3) uninstall previous version (in auto or manual)
4) install new version
5) run new version (on auto or manual)
6) open start screen
7) open main screen
8) immediate crash report.

**Screenshot(s)**
I did not take a screenshot, but I did send in 4 crash reports.

**Actual results**
Cura closes when the crash report is send or closed.

**Expected results**
Cura opens

**Project file**
Never got to the point where I could load anything

**Log file**
There are no log files for 4.4 (only for 4.0 and 4.3)

**Additional information**
none

rburema, 2020-01-08:
> I did not take a screenshot, but I did send in 4 crash reports.

I could go dig through the system, but it's a lot easier if we _know_ for certain which crashes it's about (so we don't have to guess). In that light: Could you attach the log anyway? They're in `%appdata%/cura/4.4`.

larsschwegmann, 2020-01-08:
I'm having the exact same issue on macOS Catalina. 4.3.0 works fine.

Error Traceback:

```
Traceback (most recent call last):
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentCategoryModel.py", line 101, in _update
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentModel.py", line 41, in __init__
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentModel.py", line 67, in _update
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentModel.py", line 109, in _getActiveMaterials
KeyError: 'custom_material'
```
Error log:
```
Thread 0x000070000e3f0000 (most recent call first):
  File "/Applications/Ultimaker Cura 2.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000de6a000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000d967000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 549 in wait
  File "/Applications/Ultimaker Cura 2.app/Contents/MacOS/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000d464000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000cf61000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000ca5e000 (most recent call first):
  File "/Applications/Ultimaker Cura 2.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000c55b000 (most recent call first):
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 150 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000c058000 (most recent call first):
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 138 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000b9cc000 (most recent call first):
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 112 in run
[... 61 lines cut ...]
  File "/Users/ultimaker/build/4.4/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 156 in _createDialog
[... cut ...]
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/Users/ultimaker/build/env/4.4/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
```

My System Information is in the screenshot below:

[image]

rburema, 2020-01-08:
> ... the exact same issue ...

Crashes might happen for different reasons, _especially_ when different OS'es are involved. On windows alone we're chasing at least 2 different start-up crashes (introduced after windows/driver updates).

rburema, 2020-01-08:
@larsschwegmann Did you by any chance first try 4.4.0 before starting 4.4.1?

larsschwegmann, 2020-01-08:
@rburema I have back in December, although I think I removed all application data before downgrading to 4.3 again.

I'm talking about the cura folder inside Library/Application Support etc.

dcharbonnet, 2020-01-09:
> 
> 
> @larsschwegmann Did you by any chance first try 4.4.0 before starting 4.4.1?

yes

dcharbonnet, 2020-01-09:
> 
> 
> > I did not take a screenshot, but I did send in 4 crash reports.
> 
> I could go dig through the system, but it's a lot easier if we _know_ for certain which crashes it's about (so we don't have to guess). In that light: Could you attach the log anyway? They're in `%appdata%/cura/4.4`.

dcharbonnet, 2020-01-09:
> 
> 
> > I did not take a screenshot, but I did send in 4 crash reports.
> 
> I could go dig through the system, but it's a lot easier if we _know_ for certain which crashes it's about (so we don't have to guess). In that light: Could you attach the log anyway? They're in `%appdata%/cura/4.4`.

There is no log file for 4.4.  None, zero.  Would it have been erased if I erased when I agreed to "Delete the Application Data" when I uninstalled it?
If you say the word, I'll install either 4.4.0 or 4.4.1 and take a screen shot of the error and check the log immediately.

fieldOfView, 2020-01-09:
>  Would it have been erased if I erased when I agreed to "Delete the Application Data" when I uninstalled it?

Yes. 

The word.

dcharbonnet, 2020-01-09:
Ok, here goes. 

 Clicked on the 4.4.0 Install.  Agreed to uninstall 4.3 and application data. Uninstall completed successfully.  
Proceeded with 4.4.0 Install, agreed to the license agreement.  Accepted the default location and settings. (which included Arduino Drivers, Install Visual Studio 2015 Redistributable, & Open STL files with Cura.
Install window opened and seemed to proceed normally (extractions, and at about 50% an execute file). Then The "Completing" window opened.  (the green bar didn't visibly get past 50%
Left "Run Ultimaker Cura 4.4" checked and clicked Finish
Different effect this time.  The Cura window opened with the "welcome" pop up over it.  (interestingly enough I had accidentally left 4.3 running,  It was still running but I closed it.
Clicked on Get Started and Agree and Next.  Added non-networked printer (Anycubic i3 Mega) - Next and next.

It is now running fine.

I closed Cura 4.4.0

Ran the 4.4.1 install.  Same process. Same clicks. Same answers.

Opened fine and runs.  BUT: There is a "Configuration Errors" window:  "Your configuration seems to be corrupt.  Something seems to be wrong with the following profiles: * Anycubic i# Mega * Anycubid i# Mega_user.  Would you like to reset to factory defaults?  Reset will remove all your current printers and profiles"

I clicked Reset and yes.  Cura closed.  Task Manager says it is closed.  I see no recognizable related  programs in the background.  

Restarted Cura from the Start menu Icon.  4.4.1 started and answered the same question.

Seems to be running fine.

The only thing I believe I did differently this time was to go from 4.3 to 4.4.0 and then to 4.4.1  Earlier I went from 4.3 directly to 4.4.1.  The tried to install 4.4.0 and 4.4.1 again in various ways.

I don't know if it is relevant but 4.4.1 opened in a maximized window the first time (this round).  4.4.0 and 4.4.1 after the reset opened in a medium window.

Next Question: log file:  This is the .log file under 4.4
[cura.log](https://github.com/Ultimaker/Cura/files/4042294/cura.log)

There is also this zip file in the Cura directory (not 4.4)
[4.4_20200109_140241.zip](https://github.com/Ultimaker/Cura/files/4042301/4.4_20200109_140241.zip)

{holding up hands in confusion}

I hope this helps!

Thanks!!!
DC

Ghostkeeper, 2020-01-13:
I think the original crash was happening then because of something in the 4.3 configuration that it updated for the new version. Since the installer deleted that though, this configuration is lost forever. Neither NSIS nor MSI keep back-ups of the configuration files anywhere.

We can't really reproduce the problem any more then, or guess at a reason why it wasn't loading :(

dcharbonnet, 2020-01-13:
Well, thanks for looking into the problem and I'm happy to do what I can to help.

Dak0r, 2020-02-07:
Hi,
I seem to have the same problem. Crash Logs look identical and I'm also using an Anycubic i3 mega.
I still have my working 4.3 Installation. I first tried to Update to 4.4, when that failed I switched back to 4.3 and tried to Install 4,4 / 4.4.1 next to it several times, but I keep getting the start-up crash.

I always choosed to keep my profiles etc.

If you can tell me what config files you need, I can try to provide them before uninstalling everything and starting fresh

Ghostkeeper, 2020-02-11:
We'd like to see the config files of Cura 4.3. Because we think that something is going wrong for you in the upgrade to 4.4, so with your 4.3 config files we can re-run the upgrade. To get these files:
* Open Cura 4.3.
* In the application menu press Help (or Alt+H) -> Show Configuration Folder.
* You get 1 or 2 file explorer windows popping up then, depending on your operating system. In each of those windows, go one folder higher.
* Compress the folder called "4.3" and send that over. If you have multiple windows, send both of them.

larsschwegmann, 2020-02-11:
@Ghostkeeper Here is mine, I hope it helps 😄 

[4.3.zip](https://github.com/Ultimaker/Cura/files/4185856/4.3.zip)

Dak0r, 2020-02-11:
Here's mine, I also attached the 4.4 folder just in case you need it:

[4.3.zip](https://github.com/Ultimaker/Cura/files/4186446/4.3.zip)
[4.4.zip](https://github.com/Ultimaker/Cura/files/4186448/4.4.zip)

I saw that I also still have my 3.1 and 3.6 configs, which I used before I updated to 4.3 :)

Ghostkeeper, 2020-02-13:
Thanks for the help. I loaded up Lars' configuration folder. It appears that the problem occurs for him because of the Ultimaker 2 with Olsson block. In 4.4 we moved that Olsson block version to a different printer, and it's supposed to upgrade properly but apparently not. It's then crashing because it's an Ultimaker 2 that has a material profile in it.

I looked at Dak0r's folder too and found an Ultimaker 2+ extruder with a bit of a weird naming, but I didn't try loading that. Anyways, thanks for the folders. It helps a lot to track this one down. We have reproduction steps now.

Ghostkeeper, 2020-03-02:
When I try to reproduce this myself by adding an Ultimaker 2 with Olsson block though, I'm not getting a crash :(

larsschwegmann, 2020-03-02:
@Ghostkeeper yeah it's a bit strange, even when I delete all configuration folders that I have, then perform a clean install of Cura 4.4, it still crashes...

konskarm, 2020-03-02:
I tried it in master and it does not crash anymore. It seems that it was gradually fixed. In 4.4 it was crashing on startup, in 4.5 it does not crash on startup, but crashes when creating a new profile, and in master it doesn't crash at all, even when creating a print profile. 
I noticed though that in master it recognizes the printer as Utlimaker 2 without the Olsson block. In this case if you need to transfer your print profile, save it with an Ultimaker 2, add the Ultimaker 2 with Olsson block and reload your custom print profile.

closed 2020-03-02 by konskarm