Ultimaker/Cura #6927: Allow the layer heights for the infill and support material to be 1.5 times that of the wall layer height

nathanielstenzel, 2020-01-06:
If you need the infill and support material to be a multiple of the wall layer height, you would not be able to do 0.3mm infill layer heights with 0.2mm wall layer heights. Below is an explanation of how to do this. When mixing layer heights, you may sometimes be forced to do infill or walls first, but that should not be a big problem.

start
.0 .0
add wall
.0 .2
add infill
.3 .2
add wall
.3 .4
add infill and then add wall
.6 .6

Ghostkeeper, 2020-01-13:
There is an architectural problem with this feature:

Which layer does the infill belong to? In order to draw walls and skin, we need to know where the infill is in order to implement something like Combing Mode: Infill Only, which needs to plan a travel move only through the infill. Similarly, when planning the infill moves it needs to make travel moves and avoid hitting the walls while making those travel moves.
We have the architecture that each layer has a layer plan which gets filled with lines that draw the infill, skin, walls, support, etc. If the infill and the walls are no longer in the same layer, Cura won't see it any more and won't do collision avoidance any more.

The feature request itself is valid, of course.

nathanielstenzel, 2020-01-13:
The infill and support material may need to be considered layers of all layers that it shares a Z height with.  I do not mean just the nozzle height here either but the vertical spaces where filament is. I do not know how strings from combing will affect the placement of infill below it so maybe even infill that is not placed yet should be considered. I am a bit fuzzy on how combing works at the moment though.

nathanielstenzel, 2020-01-13:
When the infill layer height is 1.5 times that of the wall layers, it will be a member of 2 wall layers.

Ghostkeeper, 2020-01-16:
We've discussed this with the team, but decided that the effort to implement it is not worth the gain in productivity/quality, considering our other priorities now. We'll keep the issue open but we won't be able to spend time on it soon.

nathanielstenzel, 2020-01-16:
That is understandable and I thank you for giving it your consideration.

I hope that actually ensuring support material under layers of support is one of the priorities that you have. That is kind of a bit of an issue. I am sure someone else has posted about it before though. If not, let me know and I will see if I can come across it happening and make a bug report on that. It probably only happens to me because I play around with settings so much.

Ghostkeeper, 2020-01-17:
> I hope that actually ensuring support material under layers of support is one of the priorities that you have. That is kind of a bit of an issue. I am sure someone else has posted about it before though.

The issue with Minimum Support Area causing some support to float in mid-air is quite high on our priority list, yes.

closed 2023-07-13 by github-actions[bot]