Ultimaker/Cura #6928: [master] AttributeError: 'CuraApplication' object has no attribute 'getHttpRequestManager'

thopiekar, 2020-01-06:
Just found this one today. Any hints how it comes?
Is there anything I need to patch, so Cura keeps running in my PPA?

```
2020-01-06 19:28:47,027 - DEBUG - [MainThread] UM.Controller.setActiveView [91]: Setting active view to SolidView
2020-01-06 19:28:47,033 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._createDialog [71]: Creating dialog [MoreInfoWindow.qml]
2020-01-06 19:28:47,104 - DEBUG - [MainThread] Toolbox.src.Toolbox._makeRequestByType [559]: Requesting [updates] metadata from server.
2020-01-06 19:28:47,104 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [114]: No auth data to retrieve the access_token from
2020-01-06 19:28:47,131 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [62]: An uncaught error has occurred!
2020-01-06 19:28:47,132 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2020-01-06 19:28:47,132 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/Toolbox/src/Toolbox.py", line 221, in _onAppInitialized
2020-01-06 19:28:47,132 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]:     self._makeRequestByType("updates")
2020-01-06 19:28:47,132 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]:   File "/usr/lib/cura/plugins/Toolbox/src/Toolbox.py", line 565, in _makeRequestByType
2020-01-06 19:28:47,132 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]:     self._application.getHttpRequestManager().get(url,
2020-01-06 19:28:47,132 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: AttributeError: 'CuraApplication' object has no attribute 'getHttpRequestManager'
2020-01-06 19:28:47,143 - DEBUG - [MainThread] ModelChecker.ModelChecker._createView [109]: Creating model checker view.
2020-01-06 19:28:47,145 - DEBUG - [MainThread] ModelChecker.ModelChecker._createView [118]: Model checker view created.
2020-01-06 19:28:47,352 - DEBUG - [MainThread] cura.CuraApplication.run [806]: Booting Cura took 6.766977071762085 seconds
Model size of -18 is less than 0
2020-01-06 19:28:47,383 - INFO - [MainThread] FirmwareUpdateChecker.FirmwareUpdateChecker.checkFirmwareVersion [74]: No machine with name BQ Hephestos 2 in list of firmware to check.
2020-01-06 19:28:47,387 - INFO - [Thread-4] FirmwareUpdateChecker.FirmwareUpdateCheckerJob.run [91]: You have a(n) Ultimaker S5 in the printer list. Do firmware-check.
2020-01-06 19:28:47,435 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [34]: Updating CustomQualityProfilesDropDownMenuModel.
2020-01-06 19:28:47,445 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [63]: Updating QualityProfilesDropDownMenuModel.
```

ChrisTerBeke, 2020-01-06:
Did you pull the latest Uranium master commits?

rburema, 2020-01-08:
This probably has to do with the 'Unify all HTTP request handling' ticket the team is working on. Latest merges to master where yesterday. Is this still a problem in master for you?

thopiekar, 2020-01-08:
@ChrisTerBeke Yes, I did (a build system for Ubuntu/Debian packages does on daily basis).
@rburema Let me check.. The package have been build 8 hours before the last time and after updating Cura at this moment... It is sadly broken still.

Looks to me like someone merged changes on the Toolbox only and missed her/his changes on CuraApplication. But yeah, no idea what really happened. Just guessing of course.

LipuFei, 2020-01-08:
@thopiekar `getHttpRequestManager()` is not in `CuraApplication`. It's in `Application`.

thopiekar, 2020-01-09:
Hmm, so looks like my Uranium builds are stuck?

closed 2020-01-11 by thopiekar