Ultimaker/Cura #6929: Skirt Height Selection

g01d10x, 2020-01-07:
I am using Cura to slice for my Ender 3, and am really happy with the performance :)

The other day for comparison I tried using another slicer, which gave the option of adding layers to the skirt for the purpose of making it easier to remove. I found that a 2 layer skirt was much easier to remove vs the single layer, especially if I've had the initial layer height set to squish the skirt a bit.

I've tried with the skirt and draft shield set to the same location, but it's not quite as effective. I would expect anyone using an extrusion based machine would benefit from this feature.

rburema, 2020-01-08:
Sorry, we've chosen to defer this for now. Adhesion is only done for the first layer right now. Extending it to the second layer (and/or beyond) would take more work than we think this is worth.

(In general, but not applicable to this situation: Instead of a skirt, you could try to 'enable prime blob', but I see that setting is disabled for Ender 3 machines by default.)

MarekPikula, 2020-08-25:
It's by default enabled in Prusa Slicer (3 layers) and it's really easy to detach after print, especially that Prusa i3 mk3(s) has magnetic bed. You basically don't even need to detach the skirt manually, because once you bend the bed, the skirt detaches by itself, because of additional stiffness.
It'd be very nice to have this option implemented. What exactly is the problem to extend it beyond the first layer @rburema?

As a workaround I have Skirt disabled and I'm using _Draft Shield_ setting with limited height to 0.5mm. It's of course different shape, as it's taking into account entire "shade" of the model, not only the first layer, but works just as well as in Prusa Slicer.

Ghostkeeper, 2020-08-27:
There isn't really a problem with it other than that the value over cost isn't very high. A number of factors contribute to this:
* We'd need to prevent the skirt from colliding with higher layers. This would be a similar implementation as the draft shield, I'd imagine.
* The intended use for the skirt is to keep distance from the model (default is 3mm) so removing the skirt isn't a problem then.
* Ultimaker printers use glass build plates which can't be bent. Not without damaging the build plate anyway.

None of these are blocking, but they do contribute to us not giving this a lot of priority over the other 1000 features requested ;)

annovialb, 2020-11-06:
Hello,
  I am casting my vote in support of this feature request.
I also have problems removing the skirt. Even using a scraper part of the skirt remains and it affects the next prints sticking to them. This is quite annoying when changing color from print to print.

I understand that avoiding the risk of collision with higher layers would add complication.
I think this feature disabled by default would be very useful even with just a warning the collisions are not avoided and should be avoided by using a low enough number of layers.

Just my 2 cents.

dougalls, 2021-01-02:
Hello.
It would also be useful to be able to set multiple layers for brims, to prevent warping materials such as ABS from breaking away from the brim at corners. This does affect glass beds as well as flexible. Slightly different issue but perhaps a similar change to the requested thicker skirt so worth considering together?

daniel-starke, 2021-02-17:
I also vote up for this feature as I am really struggling with skirt removal on my printer. Using at least two layers does really improve this situation for me using a Snapmaker 2.0 A350 with magnetic bed.

rburema, 2021-02-24:
I see that these possible workarounds have not been mentioned yet, so I'll do it now:
 - The TabAntiWarping plugin, available through the Marketplace button. (Despite what the name 'Marketplace' suggests, these plugins are very often free and open, and do usually not require login. This one is!)
 - Setting the 'Brim Distance' setting higher than 0.0mm (which is the default) can help when the brim sticks too much to the object. (Too high and you just have a sort of skirt though, of course).

5axes, 2021-11-14:
Using a post-processing script  as temporary solution : [MultiBrim.py](https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/MultiBrim.py)

eacxx, 2021-11-24:
One more vote my friends.
I don't know for others, but for me it's veeery useful this feature.

Favmir, 2021-12-31:
Would really love this feature. Single layer skirt sticks to the bed super firmly and most of the time tear apart when you try to peel it off.

Ali229, 2022-06-19:
Need this feature! Just got done with a print in Cura where I had to scrape the bed to take off the skirt, never had this problem with prusa 2 layer skirt.

Heretic8, 2022-12-12:
I stumbled upon this looking for this basic feature. 

Surprised to see this hasn't been implemented yet. Seems to be a ton of threads on it everywhere.

Phil1988, 2023-02-02:
You are right guys... this feature is very handy especially for buildplates and/or material that stick very well to the bed.

+1 from my side as not having this feature prevents me to print ASA/ABS with Cura

morremeyer, 2023-09-16:
I think this can be closed, this is available in 5.4.0. You need to set the settings selection to "Expert" or "All" to see it.

[image]

closed 2023-09-19 by rburema