Ultimaker/Cura #6931: [4.4.1] Entire print shifting during print

str3l0ck, 2020-01-07:
**Application version**
4.4.1

**Platform**
Windows 10, GTX 1080

**Printer**
Custom Printer

**Reproduction steps**
Made the part in fusion 360 and exported to stl

[TromboneStandTop.zip](https://github.com/Ultimaker/Cura/files/4029178/TromboneStandTop.zip)

Sliced the attached stl and printed it

I've resliced the file and tried again and got the same result.

**Actual results**
See attached photo

[image]

rburema, 2020-01-07:
We can't tell _that_ much from an unsliced stl, especially if the printer is 'custom'. Could you send us the project-file (saved, not exported) and/or the resulting gcode (save to file)?

nallath, 2020-01-09:
Did you get the shift at _exactly_ the same height? It looks more like a pulley of your printer getting loose (and causing the shift)

str3l0ck, 2020-01-09:
I didn't save the project file but I might still have the gcode on the SD card. I thought that my printer might be the issue but I checked and everything _seems_ ok. I had this happen twice with the same gcode and I _think_ it was the same height but I didn't measure it. For what it's worth I had this happen on another part too but for the life of me I can't remember what it was to try again or send in. I printed a few larger prints between that initial one and the 2x I tried this one (mpcnc parts, so much taller than this) and had no issues at all with those. That's what made me think it might be a slicer error.

mahtDFR, 2020-01-10:
Tested "CFFFP_Trombone Stand Top.3mf" and layer view preview in 4.4.1 works well, although this is only simulation.

By the look of it as @nallath said it's a mechanical fault with your printer, either a loose pulley, belt shift, or build plate shift mid-print.

Can you verify by printing the same stl twice using fresh gcode each time? If the issue occurs in the same place then that might indicate a software issue.

Ghostkeeper, 2020-01-14:
> Can you verify by printing the same stl twice using fresh gcode each time? If the issue occurs in the same place then that might indicate a software issue.

This could help. But even then, it could be that this same place happens to have the acceleration in just the wrong direction.

I've written an article about how to prevent layer shifts here: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/layer_shift.md

I suggest you try the things listed there under "prevention". I think this is not really a problem of Cura doing anything wrong, so we can't really fix it. It's more of a problem with the hardware and the tuning of the Cura settings to that hardware.

closed 2020-01-14 by Ghostkeeper