Ultimaker/Cura #6933: [4.4.0] Pause at height not inserting code

ed-craft, 2020-01-07:
**Application version**
4.4.0

**Platform**
Windows 10 x64

**Printer**
Creality Ender 3

**Reproduction steps**
1. Import STL file
2. Select EXTENSIONS > POST PROCESSING > MODIFY G-CODE
3. Select ADD A SCRIPT
4. Select PAUSE AT HEIGHT
5. Modify as necessary
6. SLICE the object
7. Search the gcode file using Notepad for "pause"
    a. Nothing returned
8. Print file

**Screenshot(s)**
[image]
[image]

**Actual results**
Object prints completely without pause

**Expected results**
Print should pause at defined height

**Project file**
[TestPauseAtHeight.txt](https://github.com/Ultimaker/Cura/files/4031572/TestPauseAtHeight.txt)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

ed-craft, 2020-01-07:
This apparently works when using "Pause at LAYER" but not "Pause at HEIGHT"

[TestPauseAtLayer.txt](https://github.com/Ultimaker/Cura/files/4031600/TestPauseAtLayer.txt)

rburema, 2020-01-08:
Thanks for the report. This is already fixed in master, the fix will be in the upcoming 4.5

See #6634 for more discussion.

closed 2020-01-08 by rburema