Ultimaker/Cura #6938: [4.4.1] Incorrect line width when layer height is thicker than normal

Block137, 2020-01-08:
**Application version**
4.4.1

**Reproduction steps**
1. Set "Support Infill Layer Thickness" to multiple of Layer Height
2. Enable Support Interface with 100% density and Concentric pattern

**Screenshot(s)**
[image]
Pic1: 0.08 Layer Height , 0.45 Support Line Width , **0.24 Support Infill Layer Thickness**

[image]
Pic2: 0.08 Layer Height , 0.45 Support Line Width , 0.08 Support Infill Layer Thickness

Note that in both pictures, Line Width is 0.45mm

**Actual results**
There is a gap between Support Interface and Support Infill because Support Infill is treated as multiple times wider than actual Line Width.

**Expected results**
Support Infill Line Width treated as regular Line Width
Will appear to overlap with support interface in layer view
[image]

**Additional information**
Possibly the same cause as https://github.com/Ultimaker/Cura/issues/5345

rburema, 2020-01-12:
Have you tried printing this? I think it might be a visual issue: If I take a dual extrusion printer, and slice with a different support material, then try to reproduce this, I should be able to get wildly different estimates for the length of the support-material used, but I'm not.

This would make (some) sense because, AFAIR, the line width in the preview is calculated from the amount of material extruded, which is more when it's for > 1 layers (relative to the 1 of the other types at least). In other words, those lines _would_ approximately as thick if they where to be printed on each z that is visited, but they are not.

Still mostly a bug though, even if I'm correct, just in a different system (the preview one).

Block137, 2020-01-13:
I tried and got issue with PVA support interface not adhering to PLA support infill (I have 3 extruders).

I don't think it's just visual because after #5345  got fixed, there was visible difference in preview. I thought it would have fixed both infill and support but only infill gap problem got fixed.

I got around this by reducing line width to a third and 300% flow and some crazy offset to compensate walls.

Block137, 2020-01-13:
> Have you tried printing this? I think it might be a visual issue: If I take a dual extrusion printer, and slice with a different support material, then try to reproduce this, I should be able to get wildly different estimates for the length of the support-material used, but I'm not.

The issue is not material used, just the actual gap created by mistaken line width.

rburema, 2020-01-14:
Hm... I just had a look at the fix that was applied, and it might be the reason this is happening, as the fix (unlike the original code) references the (global) layer-height and the sparse thickness specifically for infill, even though the algorithm is also used for support.

rburema, 2020-01-14:
Devs: It's on our backlog/internal system as CURA-7104 now.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

Block137, 2024-11-22:
It is working as expected in 5.8.1
[image]

closed 2024-11-22 by Block137