Ultimaker/Cura #6939: Brim height setting

kyjak, 2020-01-08:
Large prints rip off the bed,even in heated chambers, due to the brim failing sometimes.

I'd like Cura to set a Brim height and to print 2 or more layers of brim along the model.

I tried a taller 1st layer, but that affects also normal print layer not only brim, not good.

All printers, any Cura version affected.

mahtDFR, 2020-01-13:
Can you give some details about the printers and Cura versions you've tried using so I can help you more effectively?
It could be solved by fine-tuning your printer's z height settings.

Ghostkeeper, 2020-01-14:
This is what the raft was designed for. Printing a brim with multiple layers makes it very hard to remove and damages the print when you do so. Instead it's better to print a raft underneath and give it some air gap so that you can remove the adhesion structure after printing.

kyjak, 2020-01-14:
I feel bad, instead of producing a script for a solution people might use in special cases like me, you try to invent why not do it. I will code the plugin myself and maybe do a PR to Cura.
As for the printer specs - I have a Prusa MK3, printing according to specs, 1st layer is fine, extruding height is low so extrusion pressure and width is high.

Twigho, 2020-01-15:
Hello,
Maybe I can suggest a little tip : 
- Put the option "skirt/Brim flow" to  120% 
- Create a Draft Shield very close to the model with a custom height

mahtDFR, 2020-01-15:
Our expectation is that a brim with multiple layers will be hard to remove. If you are able to show us the results of your tests please share with us and we can reconsider.

Twigho, 2020-01-15:
just an idea like that 
Maybe an option that generates a support block on the brim. (separate of current support)
I will explain with an example :
With materials that are subject to a high retractation, I sometimes use an imported outline as support. Which gives a thing like that :
[image]

We can play with the options support join distance and support x/y distance (I used an the half of nozzle size)
On bottom is a support join distance 5mm
[image]
and here is 10mm
[image]

It had worked well for me but I couldn't use the support properly for print hard angles. Even using "per object setting"
It had not been difficult to remove, only a few traces due to overextrusion or poor accuracy.

kyjak, 2020-02-15:
Sorry for the delay, I simulated a 0.4mm tall brim to check how it can be removed.
There is no problem doing that, the normal way a 1 layer brim is removed - scalpel.
I attached "lily pads" to the corners to simulate a taller brim structure.
If I can suggest, a cool option for Cura would be to raise the brim on corners as an option
[image]

mporter88, 2021-03-23:
I believe that an option that created a brim that was an extension of the bottom layers. and allowed tapering down to one layer, and REQUIRED removal by mechanical methods would be a valuable addition to cura.
Such tools are not that exotic.  Certainly not in a production operation.

Rafts do not do the same thing for me as creating a larger footprint does.  Difficult materials
will pull from a raft.

I submitted this at pull #9426 with a detailed argument for the option... and it was labelled a duplicate.

vrysmlanml, 2021-06-07:
I respectfully disagree with your team. I print/cnc nylon professionally. The skirt/brim fails in Cura when printing unfilled nylon. Heated chamber or not, nylon will warp spring steel off a magnetic surface, it will tear a brim-line right at the edge of a printed part. Adding 3-5 brim or skirt layers holds the part securely to the build plate. A raft will not hold the part to the build surface and it is also too difficult to separate the part from the raft afterwards. Currently only PrusaSlicer and Simplfy3D have this feature and is what I am forced to use.

usernametakenaaaa, 2022-06-01:
Same here, I print with quite a few more difficult materials than PLA and having a thicker or thinner brim would be required.
Sometimes I go with a brin thinner than 0.5mm, sometimes I'd like it thicker than the model.
In ANY case I would not want the model to have the same height

closed 2023-07-13 by github-actions[bot]