Ultimaker/Cura #6940: Cannot Slice (Conflict with OpenVPN Client)

savante21, 2020-01-08:
**Application version**
(The version of the application this issue occurs with.)
Ultimaker Cura 4.4.1
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
MacOS MacBook Pro 16"

**Printer**
(Which printer was selected in Cura?)
Creality CR-10

**Reproduction steps**
1. Establish an OpenVPN connection using VPN Tracker 365 to my home network (where an Octopi Print Server Lives)
2. Try to slice an STL file in Cura

**Actual results**
Cura will never being slicing.  It will show that it is attempting to slice, but no progress is made.
Logs state there is an issue connecting to backend.  

** Screenshot **
[image]

**Expected results**
Expect Cura to slice the STL file.

**Project file**
https://www.dropbox.com/s/9m54y4bl4yb08js/CameraCaseMount.stl?dl=0

**Log file**
`2020-01-07 13:02:55,116 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/transparent_object.shader]...
2020-01-07 13:02:57,654 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [225]: Starting to slice...
2020-01-07 13:02:57,654 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [243]: Going to slice build plate [0]!
2020-01-07 13:02:57,662 - WARNING - [Thread-4] UM.Decorators.deprecated_function [19]: <function GlobalStack.extruders at 0x600004b14f48> is deprecated (since 4.4): Please use extruderList instead.
2020-01-07 13:02:58,110 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [429]: Sending slice message took 0.45583605766296387 seconds
2020-01-07 13:02:58,120 - DEBUG - [Thread-21] UM.Backend.Backend._backendLog [101]: [Backend] [ERROR] Could not connect to the given address
2020-01-07 13:02:58,120 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [829]: Backend quit with return code 0. Resetting process and socket.`

**Additional information**
When the OpenVPN connection is closed, Cura resumes slicing without issues.
Also, I have noticed in the logs, that Cura seems to attempt to connect to random ports on 127.0.0.1, not the same port every time.  The port doesnt matter, the slice will fail as long as there is an established openvpn connection.

savante21, 2020-01-08:
I was able to work-around this by disabling DNS requests going to remote VPN network as well as exempting 127.0.0.1 from vpn traffic.  It appears as though this issue may have been caused by an over aggressive VPN configuration/profile.  In any case, the information is out there in case it helps.

nallath, 2020-01-09:
Cura is indeed using a local socket to connect to the engine. So if the trafic from the GUI to the engine is suddenly routed via a VPN, it makes sense that it's failing.

closed 2020-01-09 by nallath

fieldOfView, 2020-01-09:
> exempting 127.0.0.1 from vpn traffic

Why would 127.0.0.1 NOT be exempted from VPN traffic? 127.0.0.1 by definition is the local computer, so traffic from it to it should never even get close to the VPN.

LipuFei, 2020-01-09:
Like @fieldOfView said, 127.0.0.1 is usually an independent network device (loopback) and the VPN connection is usually another, so VPN should not interfere if you want to access 127.0.0.1.

savante21, 2020-01-10:
Im not sure why the QVPN/OpenVPN profile was setup as aggressively as it was or why VPN Tracker wanted to route all DNS to the remote network.  I'm not exactly sure if exempting 127.0.0.1 fixed the issue or not routing all DNS to the remote VPN network fixed the issue.  In any event, as aggressive as the VPN configuration was, I only noticed issues with Cura which is why I originally thought this was a Cura specific bug.

Ghostkeeper, 2020-01-14:
Well it's nice to keep this as a workaround if someone else has a similar problem anyway :)