Ultimaker/Cura #6941: [4.4.0]Stuck on Welcome screen

lestofante, 2020-01-08:
**Application version**
4.4.0

**Platform**
arch linux, amdgpu

**Printer**
none

**Reproduction steps**
1. open cura
2. the welcome screen appear, but nothing can be done

**Screenshot(s)**
[image]

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4037229/cura.log)

smartavionics, 2020-01-08:
Your system is missing a couple of python modules. try installing trimesh and zeroconf. As you are not using the UM build you need to get whoever built your copy of cura to ensure that the required module dependencies are satisfied.

fieldOfView, 2020-01-08:
Alternatively, use the AppImage provided by Ultimaker.

smartavionics, 2020-01-08:
> Alternatively, use the AppImage provided by Ultimaker.

Good idea. Even better, use the AppImage provided by me!

nallath, 2020-01-09:
I'm closing this, since we only support the appimage.

closed 2020-01-09 by nallath