Ultimaker/Cura #6943: skirt (build plate adhesion) resolution

maukcc, 2020-01-09:
When a skirt is made around a circle the resolution is very high, resulting in very small lines of gcode that generate moves so small that they are joined in firmware resulting in shakey moves.
could there be an option to choose the resolution, or to lower it altogether?

rburema, 2020-01-12:
Yes, there are the 'maximum deviation' and the 'maximum resolution' settings. (Note that you may need to adjust _both_ of them for best results!)

maukcc, 2020-01-13:
Changing maximum deviation and/or maximum resolution results in a neglect able change in skirt.
We went from 0.01 to 1 on both and change in skirt is 0.0003mm.
Any other settings?

rburema, 2020-01-14:
I see the problem, those settings are applied before the skirt is generated, then not applied to the skirt (or brim) itself.

maukcc, 2020-01-14:
So the minimal change in skirt is probably due to repositioning of the skirt.

rburema, 2020-01-14:
Yes exactly, the model changes, due to the minimal res./dev., and then the skirt has to take a different route.

Ghostkeeper, 2020-01-14:
Since the skirt and brim are offset with a circular joint, it's going to introduce a lot more segments there.

rburema, 2020-01-14:
Fixed with https://github.com/Ultimaker/CuraEngine/commit/7bfeadfeb37b4147503c6549f0e4321cfc31a005

closed 2020-01-14 by rburema