Ultimaker/Cura #6946: brim/raft: printbed clearance border has infinite height

susisstrolch, 2020-01-09:
Application Version
Cura 4.4.1 - Ultimaker_Cura-4.4.1.AppImage

Platform
Linux / AppImage

Printer
Velleman Vertex Delta K8800

Steps to Reproduce
Try to slice a convex cylinder, using brim or raft. Lower / Upper diameter of the object is inside the clearing margin, but middle of the barrel is between inner and outer clearing margin.

Actual Results
Slicing is not possible, because the projection is larger than the inner clearing margin.

Expected results
The height of the clearing area should be either brimheight+1 or raftheight+1, so it's possible to slice such objects.

Problem will occure with any printer when the projection of the object is hitting the clearing area.

Ghostkeeper, 2020-01-14:
Yup, this is a long-standing issue since the very first version of Cura. It's caused by a limitation in our architecture where we're only able to do 2D collision checking. We don't do 3D collision checking right now.

Ghostkeeper, 2020-01-15:
We've decided to defer this issue. It's too complex, architecturally fragile and computationally expensive to implement. We'll keep this open in case we get more time or more priority for this issue.

tolookah, 2021-08-06:
I know this is closed, but what about a feature to disable the brim for the collision check, but keep it in the slicing?  some sort of advanced "brim is under an overhang, don't use it for sizing" mode? 
Or for a more difficult to implement idea, maybe a feature to dynamically adjust the brim width, if the brim is close to the edge? 

Personally, I like the first one, it sounds easier to implement.

Ghostkeeper, 2021-08-17:
We currently don't check during slicing whether the brim exceeds the build volume. This is checked before slicing, and the slicing engine doesn't repeat that check.

Checking that from CuraEngine would have two big consequences which are going to be pretty hard to solve:

* We'd need to implement some way for CuraEngine to be able to "fail" a slice, and then give feedback on why it failed. Currently CuraEngine should never fail to slice; it should always give g-code output if it starts a slice. To be able to fail a slice, it would need to be able to communicate to the front-end why the slice failed so that the front-end can display an appropriate message to the user what they need to do to fix the slice. We could purposefully divide by zero or something to crash the slicing engine but that wouldn't give any feedback to the user as to how to make their print succeed.
* This feedback would only appear halfway through slicing. The brim is generated when the progress bar of slicing is at about 50%. It is rather annoying to have to wait 2 minutes to see whether you placed all of your objects correctly. This issue is only a problem with very big models, but very big models also take the longest time to slice.

tolookah, 2021-08-18:
I might have been misunderstood/I worded it poorly.  I meant to disable the brim as a whole for the check, but have the brim enabled during slice.  A very dangerous checkbox that skips the collision check for the brim entirely.  

I appreciate your answer though, it gives me a look into what the code does, and it's limitations.

For others looking, there is a plugin that allows you to place brim pieces where you'd like, and that gets most of the way to a solution.

closed 2023-06-24 by github-actions[bot]