Ultimaker/Cura #6949: Restrict access to Ultimaker

gofh, 2020-01-10:
**Is your feature request related to a problem? Please describe.**
In a networked enviroment anybody can access the printer, send jobs, cancel running jobs, etc. That is a *big* problem

**Describe the solution you'd like**
In the web frontend there should be a setup for limiting access from certain IPs like any contemporary network printer has nowadays.

**Describe alternatives you've considered**
none

**Affected users and/or printers**
Everyone who has more than one user in their network.

**Additional context**
(Add any other context or screenshots about the feature request here.)

fieldOfView, 2020-01-10:
That would be a feature of the printer firmware, not of Cura. But since there is no public issue tracker for the printer firmware, I would not know a better place to request this.

mahtDFR, 2020-01-10:
> That would be a feature of the printer firmware, not of Cura. But since there is no public issue tracker for the printer firmware, I would not know a better place to request this.

fieldOfView is correct here, printer management takes place in firmware; and there's no dedicated place to express firmware feature requests.

Rest assured the firmware team is aware of these concerns and working on solutions.

As it's not something that can be solved in Cura I'll close the issue.

closed 2020-01-10 by mahtDFR