Ultimaker/Cura #6951: Fan speed sent to printer on wrong fan#

kornshadow097, 2020-01-11:
**Application version** Newest version 4.4.1
(The version of the application this issue occurs with.)

**Platform**windows 10
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer** tevo tarantula
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
just print as usual

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
the printer never starts the fan, it seems to set the fan speed on fan 2 which doesnt exist... it used to always set it to fan 1 which is default
**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

kornshadow097, 2020-01-11:
[TT_Pentdant Smaller Rectangle Center Cross Merge.zip](https://github.com/Ultimaker/Cura/files/4048975/TT_Pentdant.Smaller.Rectangle.Center.Cross.Merge.zip)

smartavionics, 2020-01-11:
The cooling fan number to use is set in the extruder tab of the machine settings. Here's what your project shows (sorry about the font problem, they never seem to get this right!). You can see that the value looks like 1 when it should be set to 0 for almost all printers. Hope this helps.

[image]

kornshadow097, 2020-01-11:
> The cooling fan number to use is set in the extruder tab of the machine settings. Here's what your project shows (sorry about the font problem, they never seem to get this right!). You can see that the value looks like 1 when it should be set to 0 for almost all printers. Hope this helps.
> 
> [image]

Oh ok, thanks for the help. 

So updating screws up my machines settings too? I thought the updates were supposed to move some settings if I did a back up... Having to redo all my settings is a pain. 

Especially when new settings come with the update and they are set to something horrible at default... Like right now something makes my printer start printing and rather than coming down while printing the skirt it immediately drops straight down onto the bed and then travels to the starting position leaving a trail of leaking plastic which gets stuck and curls up on the skirt and may mash into my print.. 

I didn't see anything different between printer settings under my last profile and this new one... Does cura just do this weird motion now?

Thanks

Ghostkeeper, 2020-01-14:
> So updating screws up my machines settings too?

Updating has no mechanism to create new setting overrides. The default is 0. Something must have set it to one, and it must've been a user.

> I thought the updates were supposed to move some settings if I did a back up... Having to redo all my settings is a pain.

Installing new Cura versions are supposed to leave your settings as they were as much as possible. In some cases (e.g. profiles were deleted, settings were deleted) this is not possible. For most cases they are. If updating to a new Cura version ruined your settings, we'll need to know how this happened. If making/restoring a back-up of the settings changed this, likewise we'll need to know how to reproduce it. I've updated Cura versions many times and haven't seen it change the cooling fan number.

> Especially when new settings come with the update and they are set to something horrible at default... Like right now something makes my printer start printing and rather than coming down while [...]

This issue is fixed in Cura 4.5. It was not a setting that was added or anything. It was just a fix to move to the height of the new layer before moving to the starting X/Y coordinates, so that you don't hit the previous layer when making that X/Y move. Only for the first layer this is not what you want.

Geitde, 2020-11-20:
I experienced the same when updating from 4.7.latest to 4.8. 

Suddenly I found "M106 S255 P1" in my gcodes, while all the old file created with earlier versions have just the correct "M106 S255", where the P<n> is missing do to default zerio I assume.

Two of my three printers had the fan value changed to 1. 

I am using appimage on linux, so updating is an easy process for me. That is why I am sure I had all previous versions installed. The date the gcode files starting having P1, matches the Cura install/download date. Older files are correct.

I was "lucky" that I just pre-sliced models for later prints as I upgraded marlin recently and this error kicked in when printing the calibration cubes to finetune the printers and they all fell out like melted ice cubes. Of course I searched in marlin for the reason (wrong thermistor or so), until I noticed the non spinning fan

Ghostkeeper, 2020-11-24:
We have not been able to reproduce this issue yet. (That "Fixed" label was probably added because of a different issue with the start g-code of the Tevo Tarantula. It's not the original issue in this thread.)

My reproduction steps so far for this issue are:

1. Launch Cura 4.7.1 and add a first printer, Custom FFF Printer.
2. Close Cura.
3. Start Cura 4.8.
4. Load a model (1cm cube).
5. Let it slice and save the g-code.
6. Regex search the .gcode file for "M106 S\d+ P\d".
7. In Cura, in the Machine Settings dialogue, check the value of the Cooling Fan Number setting.

I find no results in step 6. In step 7 the value is 0. Both of these results are expected.

GregValiant, 2023-09-15:
Is this still an issue or can this be closed now?

closed 2023-09-29 by github-actions[bot]