Ultimaker/Cura #6954: Gradient infill

hetile-ssense, 2020-01-11:
As demonstrated by CNC Kitchen in this video, gradient infill on the XY would be a very nice (and probably easy) feature to add to Cura.
https://www.youtube.com/watch?v=hq53gsYREHU

He provided open source python script here:
https://github.com/CNCKitchen/GradientInfill

This would make cura the only slicer to support it built-in without having to hack anything.

LarsSimonsen, 2020-01-12:
[image]
The concept is brilliantly simple. 

Btw, it has already been implemented in [PrusaSlicer as an experimental build](https://www.reddit.com/r/prusa3d/comments/enjx4d/i_added_support_for_cnc_kitchens_gradient_infill/).

hetile-ssense, 2020-01-13:
I think it would be best to have it implemented as a full infill pattern otherwise it might messup Linear Advance (marlin) or Pressure Advance (klipper)

mahtDFR, 2020-01-13:
This also caught our attention this weekend. His script adjusts extrusion rather than speed, so is optimized for direct drive machines. Bowden machines running Marlin will operate a little differently and need more tuning. We will continue to investigate...

Firedrops, 2020-01-15:
That method might not work on more complicated infills like gyroid, which probably requires the whole geometry to be re-generated with smoothly varied densities.

Cugio, 2020-01-18:
Yes, this could be very useful, and of course will probably work only with selected infill type.

5axes, 2020-01-19:
For testing purpose I've converted CNC Kitchen Python script into a Cura Postprocessing plug in Script.
[image]

Unfortunalty as mentionned by @mahtDFR  ( and also in the CNC Kitchen publication )  This solution seams to be not very efficient on a machine Bowden machines running Marlin. If some one wants to make more teste on a different machine : [https://github.com/5axes/GradientInfill/blob/master/GradientInfill.py](url)

hetile-ssense, 2020-01-19:
The suggestion would be more of having this infill pattern as another type of infill without playing with the extrusion.

5axes, 2020-01-22:
> Bowden machines running Marlin will operate a little differently and need more tuning. We will continue to investigate...

Using the gradient Infill on a machine whitout a direct drive must definetly manage a speed variation to get a correct result. 
[image]

Sample test print with a Gradiant distance of 8 mm and a infill percentage of 45%
[image]

Liger0, 2020-01-25:
It would also be different between Marlin's Linear Advance users and not I guess, that has to be considered as there is a Cura LA extension. 
Also LA can be used on both bowden and Direct drive.

zbyrek, 2021-02-02:
I wonder, why you consider changing speed\flow rather than making paths closer to eachother at edges (like setting infill 100%) and make them more further apart at center (like with infill 20% or so) when generating infil, that should work without problems in every setting (bowden\direct\linear advance).

MattLogue, 2021-07-14:
6 Vectors would be nice as cura only does top, cnc kitchen does xy+/-. Sadly, the ender3 does NOT support relative extrusion which is required for some archaic reason. I could redo it for absolute units I suppose...