Ultimaker/Cura #6955: Cura 4.4.0: Custom Profile ignores material settings if changed once

bahmrockk, 2020-01-12:
**Application version**
4.4.0

**Platform**
Windows 10

**Printer**
Ender 3 Pro; unrelated

**Reproduction steps**
1. Create custom profile.
2. Change temperature fixed for that profile, save.
3. Revert temperature back to "calculated", save.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
[image]
[image]
[image]

**Expected results**
If no custom setting is given, expected behaviour is to take the material settings when those are changed/applied.

**Profiles**
These are the profiles I used:
* Material: Generic PETG (temp of 215)
* Profile: https://pastebin.com/K0e3jYnV

rburema, 2020-01-12:
I _think_ this is(/might be related to) one of the things fixed in the patch (4.4.1). To be sure, I tried to reproduce it in that version, and unless I misunderstood something about your process, it seems that bug is not there anymore.

(Though you may have to make a new profile, I think the one posted to pastebin will always revert to 205, since 'quality changes' overwrite almost everything and the 'calculated value' of `material_print_temperature` directly points to `default_material_print_temperature`, which is somehow _also_ defined in that custom profile.)

photodude, 2020-01-13:
I believe I also have a related issue to this.
I recently created a print profile for my BIBO 2 printer which does not have quality profiles (other than the default fdmprinter global quality levels) and I also have this issue. 

Added the following in the E1 values
```json
material_final_print_temperature = =material_print_temperature - 10
material_initial_print_temperature = =material_print_temperature - 5
material_print_temperature = =default_material_print_temperature
material_standby_temperature = =material_print_temperature - 30
```
but the values are not calculating once imported
profile included for reference. 
[2020 BIBO2 E1 PLA 200 0.15mm.curaprofile.zip](https://github.com/Ultimaker/Cura/files/4052043/2020.BIBO2.E1.PLA.200.0.15mm.curaprofile.zip)

bahmrockk, 2020-01-13:
I'll retest in Cura 4.1.1 tomorrow or this evening and will answer back.

Kurolox, 2020-09-13:
As of cura 4.7.1, this bug is still present.

frankvg, 2021-04-24:
As of cura 4.9.0 this bug is still present.  Works fine in 2.5.0 (yes, sad to have to go back to such an old version to make it usable). I switch filaments all the time, and having it default to always the wrong temperatures is a huge waste of time with 4.x.

Ghostkeeper, 2021-05-11:
The problem here is that the (fx) button doesn't necessarily undo the changes in a custom quality profile or any layer on top or below that. It only restores the most recent *formula*. If the material profile didn't calculate a setting with a profile, then it's also not restored to the value in the material profile.

The (fx) button as it works now is less than useful. We have plans to replace it with something that allows settings to be removed from a custom profile.

closed 2024-12-07 by github-actions[bot]