Ultimaker/Cura #6956: [4.4.1] Initial layer temperatures not working

Firedrops, 2020-01-12:
**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Set different Initial Layer temperatures

**Screenshot(s)**
[image]

**Actual results**
Prints start with both hotend and bed temperature at "bulk" values.

**Expected results**
Prints should start with hotend and bed temperature at initial layer temperatures

**Project file**
[V slotr.zip](https://github.com/Ultimaker/Cura/files/4050646/V.slotr.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-01-12:
What is your start gcode script? It looks like something that does not come shipped with Cura. The project file you attached is not a project file but a gcode file, so I can't see how you have configured your printer.

Firedrops, 2020-01-12:
```
; Ender 3 Custom Start G-code
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G28 ; Home all axes
M117 Purge extruder
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
M117 By your command!
; End of custom start GCode
```

from [here](https://gist.github.com/besser/120f0723222c70fce88605be3517dbec). Is it causing this issue?

fieldOfView, 2020-01-12:
No offense, but no it isn't. For one thing there are comments in the gcode that are missing in what you posted above, and some `T0` arguments. The one above looks correct, but it is different from what you use. Please doublecheck what you have in Cura.

fieldOfView, 2020-01-12:
For reference, this is what is in your gcode:
```
; Ender 3 Custom Start G-code
; start preheat 1
M140 S40; start preheating the bed
M104 S220 T0; start preheating hotend
M190 S40; heat to bed setting in Cura and WAIT
M109 S220 T0; heat hotend to setting in Cura and WAIT
; end preheating
```
As you can see, the comments don't match, and there are extra `T0` arguments. The comments don't matter and neither does the `T0`, but it makes me think that the actual snippet you use is different than what you posted, and lacks the `_layer_0` parts in the replacement pattern.

Firedrops, 2020-01-13:
Thanks, I re-pasted the code and now it's working fine. I probably made a mistake while looking at a few different start G-codes.

closed 2020-01-13 by Firedrops