Ultimaker/Cura #6957: [4.3 / 4.4.1] Extra/Unneeded Code Generated at Start of Print

dB-0103, 2020-01-12:
**Application version**
4.3 and 4.4/4.4.1 (Only tested these three)

**Platform**
Windows 10 / Windows 10 Pro

**Printer**
Creality CR-10S / Stock

**Reproduction steps**
1. Slicing anything

**Screenshot(s)**
[image]

**Actual results**
At the end of custom start gcode, gantry raises to Z10mm (as expected); generated code immediately drops it back to Z0.4mm. Printer moves to start of brim and then drops again to Z0.2mm

**Expected results**
At the end of custom start gcode, gantry should raise to Z10mm, and then proceed to print the part (X, Y snd Z should all move to the first coordinate at the same time).

**Project file**
Not needed, happens with any .stl file.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
Please let me know if I need any more information, this is my first time using github.
I can provide my custom start gcode if it's needed; though the issue is not caused by it, as far as I can tell.

Ghostkeeper, 2020-01-14:
The duplicate extruder resets are still there afaik. It is indeed unnecessary but also doesn't hurt much, apart from the 7 bytes extra file size. We don't consider that to be important enough to spend time on now.

The other problem, with the Z movement at the beginning of the print, should be fixed in the next release.

Ghostkeeper, 2020-01-14:
Since there are 2 issues in this issue, I'll edit the title to limit the scope to only the one remaining issue. This issue is deferred so we consider it a bug but it doesn't get the priority for us to work on it.

dB-0103, 2020-02-29:
Little update, the Z movement is still there with the latest Cura 4.5 release. I tried adding a new machine to see if it would go away and nothing changes. It's not a big deal for me but it is still an issue.

CCS86, 2020-07-06:
In regards to the unnecessary `G92 E0`:

It may be a minor thing, but I want my start gcode to include a G92 E-0.5, so that I get a little extra nozzle prime, to compensate for a small amount of ooze when I wipe off my nozzle. I want that to occur right at the start of the first printing move, and I can't think of any other way to achieve that.

It would really be nice for Cura to not arbitrarily post that code.