Ultimaker/Cura #6958: [4.3.0, 4.4.1] Hang up on a PC with multiple monitors

Origami30, 2020-01-13:
**Application version**
Issue tested on: 4.3.0, 4.4.1

**Platform**
Windows 10 x64

**Printer**
Custom/Marlin

**Reproduction steps**
1. Install Cura and make initial setup
2. Move main window to a secondary (non-main) display
3. Close Cura
4. Start Cura --> hang up

**Screenshot(s)**
Screenshot would look like just white picture in Full HD resolution

**Actual results**
Cura main window tries to open but immediately hangs up with white screen consuming a lot of CPU

**Expected results**
Cura main window opens normally

**Project file**
Not applicable

**Log file**
Last lines:
`2020-01-13 03:09:35,478 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [111]: Initialized OpenGL subsystems.
2020-01-13 03:09:35,479 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: OpenGL Version:  4.1.0 NVIDIA 441.66
2020-01-13 03:09:35,481 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Vendor:   NVIDIA Corporation
2020-01-13 03:09:35,483 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Renderer: GeForce GTX 780/PCIe/SSE2
2020-01-13 03:09:35,485 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: GLSL Version:    4.0.0
2020-01-13 03:09:35,486 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\default.shader]...
2020-01-13 03:09:35,490 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\selection.shader]...
2020-01-13 03:09:35,492 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\select_face.shader]...
2020-01-13 03:09:35,495 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\default.shader]...
2020-01-13 03:09:35,498 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\composite.shader]...
2020-01-13 03:09:35,501 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\overhang.shader]...
2020-01-13 03:09:35,505 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\striped.shader]...
2020-01-13 03:09:35,508 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\transparent_object.shader]...
2020-01-13 03:09:35,511 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\striped.shader]...
2020-01-13 03:09:35,515 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\default.shader]...
2020-01-13 03:09:35,517 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\grid.shader]...
2020-01-13 03:09:35,520 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Programs\3D Print\Ultimaker Cura 4.4\resources\shaders\platform.shader]...`

**Additional information**
When moving main window to a secondary display Cura sets the following value in cura.cfg:
`[general]
window_left = 1920`
If I alter this value by hand and set it to **0** Cura just opens normally on the main display.

fieldOfView, 2020-01-13:
As a workaround, you can unset the "Restore window position on start" option on the General preferences page. This will always start Cura on the default monitor.

Origami30, 2020-01-13:
> As a workaround, you can unset the "Restore window position on start" option on the General preferences page. This will always start Cura on the default monitor.

@fieldOfView Thanks for this hint, it works. Despite the property `window_left=1920` main window opens on the main display and after I close it the value of `window_left` is being reset to 0.

closed 2020-01-14 by mahtDFR

Origami30, 2020-01-14:
@mahtDFR I do not agree that this issue can be closed: the workaround provided does not solve the bug itself.