Ultimaker/Cura #6959: Amount left

Ronithes, 2020-01-13:
This is just an idea, maybe there is one already but i didn't find it.
For example, let's say i bought 1Kg PLA and i am printing it all in CURA, calculator would be able to subtract material usage values per item and delete that amount from 1000g (or any amount that I enter) This would be very helpful, then i would know how much material i have left and i would not need a filament run-out sensor, CURA could alert me when i get down to 20g ( treshold,  any minimal amount that we set). So if you have software like this please let me know where can i get it, and if there is none can it be made ?. Thank you for reading my request

Ghostkeeper, 2020-01-14:
This'll work until you replace the spool for the first time. If you swap in a spool that was already opened, Cura won't be able to track how much material is left.

Something like this used to be on the road map from the moment Ultimaker decided to put a flow sensor on the printers, but it seems to have fallen off.

mahtDFR, 2020-01-15:
This was discussed this morning and we have decided not to implement. The reason is because it will not work for printer groups with more than one printer.

closed 2020-01-15 by mahtDFR