Ultimaker/Cura #6962: Speed of light movement when "hop when retracted" is used

szepeu, 2020-01-13:
**Application version**
Cura 4.1.0

**Platform**
Windows 7 SP1 64bit

**Printer**
Custom

**The problem**
When "z hop when retracted" is used, the "hop" is made as following:
[image]
The high-light-ed (pun intended) value is 299792458 m/s, which is **the speed of light in vacuum**. I guess it is some kind of initial value in the slicer which is not set properly.

**Settings string**
```
;End of Gcode
;SETTING_3 {"extruder_quality": ["[general]\\nversion = 4\\nname = Szczep Precis
;SETTING_3 e\\ndefinition = fdmprinter\\n\\n[metadata]\\nposition = 0\\ntype = q
;SETTING_3 uality_changes\\nsetting_version = 7\\nquality_type = draft\\n\\n[val
;SETTING_3 ues]\\nacceleration_print = 800\\nacceleration_travel = 1500\\naccele
;SETTING_3 ration_travel_layer_0 = 1200.0\\nbottom_layers = 3\\nbrim_line_count 
;SETTING_3 = 3\\nbrim_replaces_support = False\\ncool_fan_full_layer = 4\\ncool_
;SETTING_3 fan_speed_0 = 30\\ncool_fan_speed_max = 100\\ncool_fan_speed_min = 85
;SETTING_3 \\ncool_min_layer_time = 15\\ncool_min_layer_time_fan_speed_max = 15\
;SETTING_3 \ncool_min_speed = 5\\nfill_outline_gaps = True\\ngradual_infill_step
;SETTING_3 _height = 1\\ngradual_infill_steps = 0\\ngradual_support_infill_step_
;SETTING_3 height = 0.6\\ngradual_support_infill_steps = 3\\ninfill_enable_trave
;SETTING_3 l_optimization = True\\ninfill_line_width = 0.6\\ninfill_overlap = =1
;SETTING_3 0 if infill_sparse_density < 95 and infill_pattern != 'concentric' el
;SETTING_3 se 0\\ninfill_pattern = lines\\ninfill_sparse_density = 30\\ninfill_w
;SETTING_3 ipe_dist = 0.4\\nmaterial_flow = 110\\nmaterial_print_temperature = 1
;SETTING_3 90\\noptimize_wall_printing_order = True\\nretraction_amount = 0.8\\n
;SETTING_3 retraction_hop_enabled = True\\nretraction_min_travel = 5\\nretractio
;SETTING_3 n_speed = 50\\nskin_overlap = =5 if top_bottom_pattern != 'concentric
;SETTING_3 ' else 0\\nskirt_brim_line_width = 0.4\\nskirt_brim_minimal_length = 
;SETTING_3 150\\nskirt_brim_speed = 20\\nskirt_line_count = 1\\nspeed_infill = 2
;SETTING_3 5\\nspeed_print_layer_0 = 15\\nspeed_support = 40\\nspeed_topbottom =
;SETTING_3  20.0\\nspeed_travel = 200\\nspeed_travel_layer_0 = 200\\nspeed_wall_
;SETTING_3 0 = 18\\nspeed_wall_x = 20\\nsupport_angle = 60\\nsupport_brim_enable
;SETTING_3  = True\\nsupport_infill_rate = 20\\nsupport_offset = 2\\nsupport_pat
;SETTING_3 tern = grid\\nsupport_roof_density = 90\\nsupport_roof_enable = True\
;SETTING_3 \nsupport_roof_height = 0.6\\nsupport_roof_offset = 2\\nsupport_roof_
;SETTING_3 pattern = lines\\nsupport_xy_distance = 0.5\\nsupport_xy_distance_ove
;SETTING_3 rhang = 0.3\\nsupport_xy_overrides_z = z_overrides_xy\\ntop_layers = 
;SETTING_3 3\\nwall_0_wipe_dist = 0.8\\nwall_line_count = 3\\nxy_offset = 0\\nxy
;SETTING_3 _offset_layer_0 = -0.2\\nzig_zaggify_support = True\\n\\n", "[general
;SETTING_3 ]\\nversion = 4\\nname = Szczep Precise\\ndefinition = fdmprinter\\n\
;SETTING_3 \n[metadata]\\nposition = 1\\ntype = quality_changes\\nsetting_versio
;SETTING_3 n = 7\\nquality_type = draft\\n\\n[values]\\n\\n", "[general]\\nversi
;SETTING_3 on = 4\\nname = Szczep Precise\\ndefinition = fdmprinter\\n\\n[metada
;SETTING_3 ta]\\nposition = 2\\ntype = quality_changes\\nsetting_version = 7\\nq
;SETTING_3 uality_type = draft\\n\\n[values]\\n\\n", "[general]\\nversion = 4\\n
;SETTING_3 name = Szczep Precise\\ndefinition = fdmprinter\\n\\n[metadata]\\npos
;SETTING_3 ition = 3\\ntype = quality_changes\\nsetting_version = 7\\nquality_ty
;SETTING_3 pe = draft\\n\\n[values]\\n\\n", "[general]\\nversion = 4\\nname = Sz
;SETTING_3 czep Precise\\ndefinition = fdmprinter\\n\\n[metadata]\\nposition = 4
;SETTING_3 \\ntype = quality_changes\\nsetting_version = 7\\nquality_type = draf
;SETTING_3 t\\n\\n[values]\\n\\n", "[general]\\nversion = 4\\nname = Szczep Prec
;SETTING_3 ise\\ndefinition = fdmprinter\\n\\n[metadata]\\nposition = 5\\ntype =
;SETTING_3  quality_changes\\nsetting_version = 7\\nquality_type = draft\\n\\n[v
;SETTING_3 alues]\\n\\n", "[general]\\nversion = 4\\nname = Szczep Precise\\ndef
;SETTING_3 inition = fdmprinter\\n\\n[metadata]\\nposition = 6\\ntype = quality_
;SETTING_3 changes\\nsetting_version = 7\\nquality_type = draft\\n\\n[values]\\n
;SETTING_3 \\n", "[general]\\nversion = 4\\nname = Szczep Precise\\ndefinition =
;SETTING_3  fdmprinter\\n\\n[metadata]\\nposition = 7\\ntype = quality_changes\\
;SETTING_3 nsetting_version = 7\\nquality_type = draft\\n\\n[values]\\n\\n"], "g
;SETTING_3 lobal_quality": "[general]\\nversion = 4\\nname = Szczep Precise\\nde
;SETTING_3 finition = fdmprinter\\n\\n[metadata]\\ntype = quality_changes\\nsett
;SETTING_3 ing_version = 7\\nquality_type = draft\\n\\n[values]\\nacceleration_e
;SETTING_3 nabled = True\\nadhesion_type = brim\\nlayer_height = 0.15\\nlayer_he
;SETTING_3 ight_0 = 0.25\\nmagic_spiralize = False\\nretraction_combing = all\\n
;SETTING_3 support_enable = True\\nsupport_type = buildplate\\n\\n"}

```

hetile-ssense, 2020-01-13:
Wasn't this already fixed?

https://github.com/Ultimaker/Cura/issues/5753

fieldOfView, 2020-01-13:
Yes, it was. But not retroactively in Cura 4.1

Ghostkeeper, 2020-01-14:
Fixed in version 4.2.

closed 2020-01-14 by Ghostkeeper