Ultimaker/Cura #6963: Mistake in wiki, configuration page

ArnaudTa, 2020-01-13:
Hello, this [page](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations#linux) in wiki could be improved.

$USER should be replaced by $HOME you don't have to explain it and the paths from wiki could be copy/paste directly in file manager.

Thanks.

rburema, 2020-01-14:
Isn't it distribution/shell dependent which one gets set?

ArnaudTa, 2020-01-14:
AFAIK it's true for all distribution/shell

Ghostkeeper, 2020-01-14:
Not in Windows apparently though: https://superuser.com/questions/607105/is-the-home-environment-variable-normally-set-in-windows

fieldOfView, 2020-01-14:
But for Windows, that page should really just use %APPDATA% instead of $USER/AppData/Roaming and %APPDATALOCAL% instead of $USER/AppData/Local. $USER is syntactically the wrong notation in Windows (not even recognised in a path), and it semantically wrong in systems that have changed the location of the roaming app data (eg stored on a network drive).

ArnaudTa, 2020-01-14:
The link leads to the Linux section. I should have precise it in report.

Ghostkeeper, 2020-01-14:
As far as I could find, for Linux indeed the $HOME seems to be available everywhere.

Ghostkeeper, 2020-01-14:
Fixed it!

Ghostkeeper, 2020-01-14:
For Windows we need to look into further improvements. Didn't anyone test that?

closed 2020-01-14 by Ghostkeeper

rburema, 2020-01-14:
The linked page is for linux only :-)

The windows one has been updated a number of times now. The latest changes where just last week when Matt needed to install from source.

mahtDFR, 2020-01-16:
Thanks for the report, investigating Windows now

mahtDFR, 2020-01-20:
Updated, if there are further comments please raise them and I can make improvements.
https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

fieldOfView, 2020-01-20:
The Windows paths should use `\` (backslash) instead of `/` (forward slash). Also the HOMEPATH that is explained is not used in the paths.

mahtDFR, 2020-01-20:
> The Windows paths should use `\` (backslash) instead of `/` (forward slash). Also the HOMEPATH that is explained is not used in the paths.

Good points, updated.