Ultimaker/Cura #6965: Cura not slicing stl after viewing gcode file

Barry37, 2020-01-13:
**Application version**
Cura 4.4

**Platform**
Mac OSX Yosemite

**Printer**
Lulzbot Taz 4

**Reproduction steps**
1. Opened a gcode file
2. Cleared build plate
3. Opened an stl file

**Screenshot(s)**
Attached image showing warning after attempted save of stl

**Actual results**
Button, bottom right says "Save to File"

**Expected results**
Button should say "Slice" for a newly opened stl file

**Project file**
Happens with any gcode file (all are generated by Cura)

**Log file**
GCodeReader.FlavorParser.processGCodeStream [303]: Preparing to load GCode
2020-01-13 15:44:33,209 - DEBUG - [Thread-2] GCodeReader.FlavorParser.processGCodeStream [343]: Parsing Gcode...
2020-01-13 15:44:35,112 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [135]: New error check scheduled.
2020-01-13 15:44:39,914 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [34]: Updating CustomQualityProfilesDropDownMenuModel.
2020-01-13 15:44:40,364 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [63]: Updating QualityProfilesDropDownMenuModel.
2020-01-13 15:44:44,149 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [187]: Error check finished, result = False, time = 9.1s
2020-01-13 15:44:50,294 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2020-01-13 15:45:05,268 - DEBUG - [Thread-2] GCodeReader.FlavorParser.processGCodeStream [473]: Finished parsing Gcode
2020-01-13 15:45:05,278 - DEBUG - [Thread-2] GCodeReader.FlavorParser.processGCodeStream [484]: GCode loading finished
2020-01-13 15:45:05,319 - DEBUG - [Thread-2] UM.FileHandler.ReadFileJob.run [71]: Loading file took 33.0 seconds
2020-01-13 15:45:05,841 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:05,849 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:05,857 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:06,064 - DEBUG - [MainThread] UM.Operations.OperationStack.push [64]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x7fd4a24918c8>, took 3ms
2020-01-13 15:45:06,168 - DEBUG - [MainThread] UM.Controller.setActiveStage [149]: Setting active stage to PreviewStage
2020-01-13 15:45:06,456 - DEBUG - [MainThread] UM.Controller.setActiveView [91]: Setting active view to SimulationView
2020-01-13 15:45:06,545 - INFO - [MainThread] STLReader.STLReader.load_file [46]: Using NumPy-STL to load STL data.
2020-01-13 15:45:06,688 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [482]: Calculating normals took 0.0010797977447509766 seconds
2020-01-13 15:45:06,690 - DEBUG - [MainThread] STLReader.STLReader._read [97]: Loaded a mesh with 12612 vertices
2020-01-13 15:45:06,704 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/SimulationView/simulationview_composite.shader]...
2020-01-13 15:45:07,222 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:07,231 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:07,240 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:07,252 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/PreviewStage/PreviewMain.qml:39:5: QML ActionPanelWidget: Binding loop detected for property "width"
2020-01-13 15:45:07,920 - DEBUG - [MainThread] UM.Controller.setActiveView [91]: Setting active view to SimulationView
2020-01-13 15:45:08,046 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/color.shader]...
2020-01-13 15:45:08,094 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/SimulationView/layers.shader]...
2020-01-13 15:45:08,128 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/SimulationView/layers_shadow.shader]...
2020-01-13 15:45:08,162 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/toolhandle.shader]...
2020-01-13 15:45:08,190 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/color.shader]...
2020-01-13 15:45:16,284 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2020-01-13 15:45:27,847 - DEBUG - [MainThread] UM.Operations.OperationStack.push [64]: GroupedOp.(#=1) <UM.Operations.RemoveSceneNodeOperation.RemoveSceneNodeOperation object at 0x7fd4a3ea8588>, took 17ms
2020-01-13 15:45:28,305 - WARNING - [MainThread] cura.UI.PrintInformation.setBaseName [357]: Unsupported Mime Type Database file extension 
2020-01-13 15:45:32,624 - DEBUG - [MainThread] UM.Controller.setActiveStage [149]: Setting active stage to PrepareStage
2020-01-13 15:45:32,632 - DEBUG - [MainThread] UM.Controller.setActiveView [91]: Setting active view to SolidView
2020-01-13 15:45:32,707 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2020-01-13 15:45:32,955 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:32,964 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:32,972 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 15:45:32,984 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/PrepareStage/PrepareMain.qml:16:5: QML ActionPanelWidget: Binding loop detected for property "width"
2020-01-13 15:46:43,284 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2020-01-13 15:46:45,788 - INFO - [Thread-1] STLReader.STLReader.load_file [46]: Using NumPy-STL to load STL data.
2020-01-13 15:46:45,884 - DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [482]: Calculating normals took 0.002830028533935547 seconds
2020-01-13 15:46:45,886 - DEBUG - [Thread-1] STLReader.STLReader._read [97]: Loaded a mesh with 30216 vertices
2020-01-13 15:46:45,937 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [439]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.03062915802001953 seconds. 28 input vertices. 28 output vertices.
2020-01-13 15:46:45,972 - DEBUG - [Thread-1] UM.FileHandler.ReadFileJob.run [71]: Loading file took 0.2 seconds
2020-01-13 15:46:46,039 - DEBUG - [Thread-1] UM.Mesh.MeshData.approximateConvexHull [439]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.05712008476257324 seconds. 28 input vertices. 28 output vertices.
[... cut ...]
2020-01-13 15:47:05,695 - INFO - [MainThread] cura.CuraApplication.resetAll [1253]: Resetting all scene transformations
2020-01-13 15:47:05,715 - DEBUG - [MainThread] UM.Operations.OperationStack.push [64]: GroupedOp.(#=1) SetTransformOp.(node=<CuraSceneNode object: 'ND51_TEEstrapping.stl'>), took 10ms
2020-01-13 15:47:24,288 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles

**Additional information**
This was also happening with 4.3 & 4.2. Every gcode file opened causes the problem. 
[image]

Can only get past this by Quitting & Reopening Cura.

Ghostkeeper, 2020-01-14:
Hmm, the log file is not saying anything about this. Does it work differently if you wait a few seconds extra between each operation?

Here's a video demonstrating what I did: [slicing_after_gcode.zip](https://github.com/Ultimaker/Cura/files/4059541/slicing_after_gcode.zip)

Barry37, 2020-01-14:
Thanks for video.

Opening an stl file from the [File} menu, or via the folder Icon at the top left hand corner (“Prepare” view) makes no difference - there is still no slicing option after viewing a gcode file, clearing the build plate and loading an stl file.
Waiting (even a few minutes) between operations makes no difference.

However, discovered today that this only applies to gcode files sliced with Cura 4.x., not 3.6.3. Viewing a gcode file sliced with the latter, in 4.4.1, does allow subsequent slicing of an stl file.
Also, after opening a 4.x gcode file (and losing Slicing option), then opening an older 3.6.3 gcode file, stl slicing ability is restored without quitting.

The same stl file sliced into gcode with 3.6.3 has more lines in the log file, when opened in 4.4.1, than when sliced with 4.4.1 and opened as gcode. - snippets attached (two files)

Barry.

Ghostkeeper, 2020-01-15:
Yeah indeed I could reproduce this with [this g-code from 3.6](https://github.com/Ultimaker/Cura/files/4065189/test.zip). It doesn't mention anything special in the logs.

Barry37, 2020-01-15:
Opening your test.gcode file in Cura 4.4.1 makes it lose the [slice] option on stl files. (perhaps that’s what you meant)

Oddly, the 4.4.1 file of the gcode pair (3.6.3/4.4.1) saved yesterday, is no longer causing loss of [slice], but other files still are.

An Activity Monitor sample of Cura has a lot of  “FAILED: dyld LOAD of…, but I don’t know the significance of these - most failed to load files do exist in usr/lib/system etc. , but some do not.

rburema, 2020-01-16:
Devs: Now in our internal system as ticket 'CURA-7110'.