Ultimaker/Cura #6966: Make Overhang Printable subtract instead of add volume

haleypearse, 2020-01-13:
"Make Overhang Printable" adds volume under overhangs, apparently by projecting downwards from the top of overhangs at the "Maximum Model Angle."

Instead I want to project up from the bottom of overhangs, removing volume from the object.

Hopefully the implementation would be a minor change from the original, and potentially just as useful, but in unforeseeable ways.

My application is printing globoid worm gears, where the overhangs are ruining the gear, but cutting away some of the mating surface would not be a problem.

[image]

https://www.thingiverse.com/make:708592

Ghostkeeper, 2020-01-14:
Interesting idea!

haleypearse, 2020-03-10:
I would argue removing material would be much more useful than the feature is currently. Circular bolt holes require exactly this, 'make printable' but removing material. It would also compensate for overhang droop, especially for small but extreme overhangs.

Ghostkeeper, 2020-03-13:
We've discussed this with the team and decided that we can't warrant spending time on this because the user base of this setting is very small. It's a catch-22, I know, but there's so much triage going on always and the real answer is that you probably have to work this into your model in the design phase anyway.

haleypearse, 2020-11-28:
Bump. Fair points, but I think will eventually be a feature in slicers and be commonly used. The problems in a model are often overhanging perimeters, both for figurines and dimensional parts, and it only takes a few to make a stringy mess. A simple way to achieve this algorithmically would be this. At each layer, move any (non-bridging) overhanging perimeter toward the model interior until it doesn't overhang. Successive layers will have an interior area deficit until the perimeter matches the model again.

By configuring the overhang threshold angle, you could get as close as possible to the original without print errors, making a lot of support-requiring or unprintable models printable. It doesn't make sense to anticipate and implement this in the design phase because, first, It's a lot harder, and second, overhang angle is subject to print parameters like layer height and nozzle size.

Ghostkeeper, 2021-07-22:
I had a feature request via Discord from someone that demonstrates the use of such a feature too. They were printing a scoop, with the spoon upside down like a dome. Their support was reliable for the flat top surface but not so reliable for slopes, due to difficulty with the Support X/Y Distance and Z distance. For that sort of stuff, making overhang printable subtract could be a solution as well.

closed 2023-07-13 by github-actions[bot]