Ultimaker/Cura #6967: [4.4.1] crashes when previewing on Manjaro KDE with radeon graphic card

jellespijker, 2020-01-13:
**Application version**
4.4.1

**Platform**

`uname -a`
Linux ####### 5.4.6-2-MANJARO #1 SMP PREEMPT Tue Dec 24 15:55:20 UTC 2019 x86_64 GNU/Linux
with KDE desktop

graphics card:
Mullins [Radeon R4/R5 Graphics](ATI Technology Inc)

**Printer**
UM 3

**Reproduction steps**
1. import STL
2. Slice
3. Press preview (both on toolbar and pop-up)

**Screenshot(s)**
Not relevant

**Actual results**
The GUI crashed

**Expected results**
Preview of the sliced model should be shown

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
```
2020-01-13 18:34:39,161 - INFO - [Thread-1] SliceInfoPlugin.SliceInfoJob.run [34]: Sending anonymous slice info to [https://stats.ultimaker.com/api/cura]...
2020-01-13 18:34:39,294 - INFO - [Thread-1] SliceInfoPlugin.SliceInfoJob.run [38]: Sent anonymous slice info.
2020-01-13 18:34:40,532 - WARNING - [MainThread] UM.Settings.InstanceContainer.setCachedValues [148]: Unable set values to be lazy loaded when values are already loaded 
2020-01-13 18:34:41,587 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition fdmprinter failed to pickle.
2020-01-13 18:34:42,794 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker3_extended failed to pickle.
2020-01-13 18:34:43,985 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker2_plus failed to pickle.
2020-01-13 18:34:45,435 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker2_extended_plus failed to pickle.
2020-01-13 18:34:46,773 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker_s5 failed to pickle.
2020-01-13 18:34:48,123 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker_s3 failed to pickle.
2020-01-13 18:34:48,494 - WARNING - [MainThread] UM.Settings.InstanceContainer.setCachedValues [148]: Unable set values to be lazy loaded when values are already loaded 
2020-01-13 18:34:50,300 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker2_extended failed to pickle.
2020-01-13 18:34:52,037 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker2_go failed to pickle.
2020-01-13 18:34:53,444 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker2 failed to pickle.
2020-01-13 18:34:54,644 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [269]: The definition cache for definition ultimaker_original failed to pickle.
2020-01-13 18:34:54,715 - DEBUG - [Thread-3] UM.FileHandler.WriteFileJob.run [71]: Writing file took 15.548999786376953 seconds
2020-01-13 18:34:54,954 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [53]: Refreshing the access token.
2020-01-13 18:34:55,250 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [208]: Attempting to store the auth data
2020-01-13 18:34:55,341 - DEBUG - [MainThread] UM.Controller.setActiveStage [149]: Setting active stage to PreviewStage
2020-01-13 18:34:55,424 - DEBUG - [MainThread] UM.Controller.setActiveView [91]: Setting active view to SimulationView
2020-01-13 18:34:55,429 - INFO - [MainThread] STLReader.STLReader.load_file [54]: Using legacy code to load STL data.
2020-01-13 18:34:55,587 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [482]: Calculating normals took 0.0019712448120117188 seconds
2020-01-13 18:34:55,589 - DEBUG - [MainThread] STLReader.STLReader._read [97]: Loaded a mesh with 12612 vertices
2020-01-13 18:34:55,609 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/tmp/.mount_Ultima5iLdez/usr/bin/plugins/plugins/SimulationView/simulationview_composite.shader]...
2020-01-13 18:34:55,755 - DEBUG - [Thread-2] CuraEngineBackend.ProcessSlicedLayersJob.run [70]: Processing new layer for build plate 0...
2020-01-13 18:34:57,632 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///tmp/.mount_Ultima5iLdez/usr/bin/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 18:34:57,726 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///tmp/.mount_Ultima5iLdez/usr/bin/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 18:34:58,075 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///tmp/.mount_Ultima5iLdez/usr/bin/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
2020-01-13 18:34:58,145 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///tmp/.mount_Ultima5iLdez/usr/bin/plugins/plugins/PreviewStage/PreviewMain.qml:39:5: QML ActionPanelWidget: Binding loop detected for property "width"
2020-01-13 18:35:00,496 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/tmp/.mount_Ultima5iLdez/usr/bin/resources/shaders/color.shader]...
2020-01-13 18:35:00,594 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/tmp/.mount_Ultima5iLdez/usr/bin/plugins/plugins/SimulationView/layers3d.shader]...
2020-01-13 18:35:00,667 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/tmp/.mount_Ultima5iLdez/usr/bin/plugins/plugins/SimulationView/layers3d_shadow.shader]...
2020-01-13 18:35:00,764 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/tmp/.mount_Ultima5iLdez/usr/bin/resources/shaders/toolhandle.shader]...
2020-01-13 18:35:00,839 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [/tmp/.mount_Ultima5iLdez/usr/bin/resources/shaders/color.shader]...
2020-01-13 18:35:05,297 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [34]: Updating CustomQualityProfilesDropDownMenuModel.
2020-01-13 18:35:05,346 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [63]: Updating QualityProfilesDropDownMenuModel.
2020-01-13 18:35:09,020 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2020-01-13 18:35:09,739 - DEBUG - [Thread-2] CuraEngineBackend.ProcessSlicedLayersJob.run [250]: Processing layers took 13.983329057693481 seconds
2020-01-13 18:35:09,848 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onProcessLayersFinished [859]: See if there is more to slice(2)...

```

**Additional information**
Core dump
```
Process 14519 (cura) of user 1000 dumped core.

Stack trace of thread 14786:
#0  0x00007f8d6dbad553 n/a (radeonsi_dri.so)
#1  0x00007f8d6dbad740 n/a (radeonsi_dri.so)
#2  0x00007f8d6dbae2f0 n/a (radeonsi_dri.so)
#3  0x00007f8d6dbae537 n/a (radeonsi_dri.so)
#4  0x00007f8d6dc80fa2 n/a (radeonsi_dri.so)
#5  0x00007f8d6dc810f1 n/a (radeonsi_dri.so)
#6  0x00007f8d6dc81854 n/a (radeonsi_dri.so)
#7  0x00007f8d6dc68568 n/a (radeonsi_dri.so)
#8  0x00007f8d6dc5dfc0 n/a (radeonsi_dri.so)
#9  0x00007f8d6e43b0ac n/a (radeonsi_dri.so)
#10 0x00007f8d6e437c20 n/a (radeonsi_dri.so)
#11 0x00007f8d6e28db0c n/a (radeonsi_dri.so)
#12 0x00007f8d6e28d658 n/a (radeonsi_dri.so)
#13 0x00007f8d86d314cf start_thread (libpthread.so.0)
#14 0x00007f8d86b032d3 __clone (libc.so.6)

Stack trace of thread 14571:
#0  0x00007f8d86d37c45 pthread_cond_wait@@GLIBC_2.3.2 (libpthread.so.0)
#1  0x00007f8d6e28da5c n/a (radeonsi_dri.so)
#2  0x00007f8d6e28d658 n/a (radeonsi_dri.so)
#3  0x00007f8d86d314cf start_thread (libpthread.so.0)
#4  0x00007f8d86b032d3 __clone (libc.so.6)

Stack trace of thread 14582:
#0  0x00007f8d86d3a704 do_futex_wait.constprop.0 (libpthread.so.0)
#1  0x00007f8d86d3a7f8 __new_sem_wait_slow.constprop.0 (libpthread.so.0)
#2  0x00000000004b6a4f n/a (/tmp/.mount_Ultima5iLdez/usr/bin/cura)

```

*I can take my laptop to work if needed*

Ghostkeeper, 2020-01-14:
Sooooo... segfault in the Radeon drivers?

Does it happen with any file or just big ones?

jellespijker, 2020-01-14:
Define big?

It happens with a **2.5 mb stl file** (converted to a **27.9 mb gcode file**), which isn't that big i.m.o..
But it also crashes with a **27 kb stl file**.

jellespijker, 2020-01-14:
Out put:

```bash
$ lscpu
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   40 bits physical, 48 bits virtual
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       AuthenticAMD
CPU family:                      22
Model:                           48
Model name:                      AMD A8-6410 APU with AMD Radeon R5 Graphics
Stepping:                        1
Frequency boost:                 enabled
CPU MHz:                         1033.000
CPU max MHz:                     2000.0000
CPU min MHz:                     1000.0000
BogoMIPS:                        3994.12
Virtualization:                  AMD-V
L1d cache:                       128 KiB
L1i cache:                       128 KiB
L2 cache:                        2 MiB
NUMA node0 CPU(s):               0-3
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Full AMD retpoline, STIBP disabled, RSB filling
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_
                                 tsc rep_good acc_power nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c lahf_lm
                                  cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt topoext perfctr_nb bpext ptsc perfctr_llc cpb hw_pstate ssbd vmm
                                 call bmi1 xsaveopt arat npt lbrv svm_lock nrip_save tsc_scale flushbyasid decodeassists pausefilter pfthreshold overflow_recov

```

```bash
$ cat /proc/meminfo
MemTotal:       15331900 kB
MemFree:        11748676 kB
MemAvailable:   13081660 kB
Buffers:           91720 kB
Cached:          1854748 kB
SwapCached:            0 kB
Active:          1832096 kB
Inactive:        1386172 kB
Active(anon):    1278672 kB
Inactive(anon):   400660 kB
Active(file):     553424 kB
Inactive(file):   985512 kB
Unevictable:          16 kB
Mlocked:              16 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:               112 kB
Writeback:             0 kB
AnonPages:       1271856 kB
Mapped:           556040 kB
Shmem:            411248 kB
KReclaimable:      91304 kB
Slab:             170920 kB
SReclaimable:      91304 kB
SUnreclaim:        79616 kB
KernelStack:       10976 kB
PageTables:        30044 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     7665948 kB
Committed_AS:    7057232 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       42148 kB
VmallocChunk:          0 kB
Percpu:             2064 kB
HardwareCorrupted:     0 kB
AnonHugePages:         0 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
FileHugePages:         0 kB
FilePmdMapped:         0 kB
HugePages_Total:       0
HugePages_Free:        0
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:               0 kB
DirectMap4k:      287488 kB
DirectMap2M:     7016448 kB
DirectMap1G:     9437184 kB
```

```bash
$ glxinfo -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: X.Org (0x1002)
    Device: AMD KABINI (DRM 2.50.0, 5.4.6-2-MANJARO, LLVM 9.0.0) (0x9851)
    Version: 19.3.1
    Accelerated: yes
    Video memory: 1024MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 4.5
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 467 MB, largest block: 467 MB
    VBO free aux. memory - total: 2012 MB, largest block: 2012 MB
    Texture free memory - total: 467 MB, largest block: 467 MB
    Texture free aux. memory - total: 2012 MB, largest block: 2012 MB
    Renderbuffer free memory - total: 467 MB, largest block: 467 MB
    Renderbuffer free aux. memory - total: 2012 MB, largest block: 2012 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 1024 MB
    Total available memory: 3067 MB
    Currently available dedicated video memory: 467 MB
OpenGL vendor string: X.Org
OpenGL renderer string: AMD KABINI (DRM 2.50.0, 5.4.6-2-MANJARO, LLVM 9.0.0)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 19.3.1
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.5 (Compatibility Profile) Mesa 19.3.1
OpenGL shading language version string: 4.50
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 19.3.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

```

Ghostkeeper, 2020-01-15:
> Define big?

I was looking for something that could cause your GPU to run out of memory. That's something that the video driver may not have caught.

Ghostkeeper, 2020-01-15:
Does it crash when you go into compatibility mode (in the Preferences screen)?

jellespijker, 2020-01-15:
No crash when in compatibility mode

[image]

Ghostkeeper, 2020-01-16:
Interesting. But I'm still blaming your video driver here! At least you have a workaround now.

Compatibility mode lowers the OpenGL version to 2.0. Otherwise Cura needs OpenGL 4 to get support for geometry shaders and the fragment sampler. Your GPU claims to support OpenGL 4.5 but there seems to be something awry there.

jellespijker, 2020-01-16:
Nice to have the work around.

But I'm not completely convinced about laying the blame at the GPU. It survived various stress test and I switched from the video-vesa driver to the video-linux driver. With the same result.

B.t.w when running the appimage from the shell, the last output is:
```
/tmp/.mount_UltimaZgdgPr/usr/bin//cura.sh: line 14:  1452 Floating point exception(core dumped) cura "$@"
```
This is not found in the log files. Is there a division by zero somewhere?

Ghostkeeper, 2020-01-17:
That shell script is what starts the Python environment.

If there is a division by zero somewhere in our front-end code, we'd get a Python ZeroDivisionError, which would get caught by the crash handler and shown to the user in a pop-up. If there is a division by zero somewhere in our back-end code, the slice would stop halfway and the front-end would survive.

The only possibility then is that there is a division by zero somewhere in one of the native libraries called by our front-end. This is not normally under our control, but it could be that we're calling one of these native libraries in such a way that they experience an internal error. There could be a way to call on these libraries differently, but it's very hard to track that down. There should not be an internal error in those libraries of course.

jellespijker, 2020-01-17:
Can I check the dependencies with a tool such as ldd, and update the likely culprits accordingly?

Ghostkeeper, 2020-01-17:
You could run the application through GDB or something like it to see where in the call stack it ends up raising that floating point exception. This would identify which dependency is the culprit.

closed 2020-06-19 by no-response[bot]