Ultimaker/Cura #6968: [4.4.1] custom profile calculations are not all calculating

photodude, 2020-01-13:
I recently created a print profile for my BIBO 2 printer and edited it to add calculations to some values. The material temperatures and top/bottom thickness or wall_thickness are not adjusting according to the values entered.

**Application version**
4.4.1

**Platform**
Win 10

**Printer**
BIBO 2 E1 only

**Reproduction steps**
1. Exported an existing custom profile.
2. edited the `.curaprofile` archive text files with 7zip
3. Added the following formula values in the e1a values  
```json
material_final_print_temperature = =material_print_temperature - 10
material_initial_print_temperature = =material_print_temperature - 5
material_print_temperature = =default_material_print_temperature
material_standby_temperature = =material_print_temperature - 30
raft_airgap = =layer_height*2
support_interface_height = =layer_height*6
support_z_distance = =layer_height*2
top_bottom_thickness = =layer_height_0+layer_height*5
wall_thickness = =line_width*3
```
4. saved the archive files with edited profile names
5. imported the new profile in cura
6. checked for expected fields to be modified acording to the formulas entered. 

**Actual results**
the values are not calculating once the modified profile is imported under a new name

**Expected results**
fields with formulas should override the field and give the calculated results

**Profile file**
[2020 BIBO2 E1 PLA 200 0.15mm.curaprofile.zip](https://github.com/Ultimaker/Cura/files/4054994/2020.BIBO2.E1.PLA.200.0.15mm.curaprofile.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4054997/cura.log)

**Additional information**
While trying to make screenshots cura 4.4.1 crashed. Error report sent, and log file included here in case it's relevant.

Ghostkeeper, 2020-01-14:
You're getting maximum recursion depth errors because of this profile. The reason for that is that in the global settings of that .curaprofile, it sets this line:

    material_bed_temperature = =material_bed_temperature

This is recursion. Look into the file `bibo2_single_e1_bibo2_e1_pla_212` in that archive (not the one you're editing).

closed 2020-01-14 by Ghostkeeper

photodude, 2020-01-14:
@Ghostkeeper  derp... total copy paste error. That line should have been as follows
```json
material_bed_temperature = =default_material_bed_temperature
```
corrected that line and resolved the issue with the calculations not working as expected. Thanks again for catching my error.

Although I had to hunt for the offending imported file in the `Users\[user]\AppData\Roaming\cura\4.4\quality_changes` folder to get Cura to stop crashing from that bad imported profile. 
it might be helpful If there was a way for Cura to identify recursion depth errors in profiles to users and to let users decide if they want to isolate or delete such profiles with problems. If I don't know where to look and for what I would have and to take the "nuclear reinstall" option to get Cura to run.

Ghostkeeper, 2020-01-15:
These things can only happen if you manually modify the configuration files, and then make a mistake. We've so far always said that if you modify the configuration files manually (with a text editor rather than through Cura) then you're basically on your own. We expect the users that do this to be technical enough to read the log file to track down the issue. Of course we're willing to make this easier to some extent because the developers also have to make such changes from time to time.

In this case I read the log file and saw the recursion error. The recursion error didn't specify which setting was causing the mistake, but in Cura it was displaying the bed temperature at 0. So I looked in the [God Mode](https://github.com/sedwards2009/cura-god-mode-plugin) plug-in to see what's up with that and there I saw that the bed temperature was defined as `=material_bed_temperature` from the global stack.

We could improve the debugability of profiles, I agree. But then again, that God Mode plug-in already does that for us.