Ultimaker/Cura #6969: I can't print at the right temperature

Nigthscorpion, 2020-01-13:
**Application version**
(4.2)

**Platform**
(Mac OSX)

**Printer**
(Anycubic I3 Mega)

**Reproduction steps**
1. (The printer that I use requires me to print the material at 245ºC, it's a basic PLA, and it will not print without the PLA at the default settings as 200ºC, so I changed the material temp to 245º, however, it will not let me print because it gets rid of the material/layer height setting)
2. (I tried manually adjusting layer size and the printing temp, it didn't work)

**Screenshot(s)**
[image]
[image]
 

**Actual results**
(I won't let me print)

**Expected results**
(I want to print.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-01-14:
Just tried replicating in 4.4.1 (Windows) with no issues. Can you update to the latest version and try again?

Nigthscorpion, 2020-01-14:
Wow, I feel quite stupid, thanks, worked like a charm

closed 2020-01-14 by Nigthscorpion