Ultimaker/Cura #6971: Scaling problem

Willem43T, 2020-01-14:
**Application version**
Cura 4.4.0 and 4.4.1

**Platform**
Windows 10
**Printer**
Ender 3

**Reproduction steps**
1. Trying to scale a model changing only X and Y leaving Z unchanged
2. Un-tick "Uniform Scaling"
3. Easiest way to observe problem is to drag the Y axis handle. Y-axis dimension changes but Z-Axis percentage changes completely distorting the model.
4. Dragging the Z-axis handle changes Z-axis dimension but changes the Y-axis percentage, again distorting the model.
5. By now everything is out of control and model has to be reloaded.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Model is unusable

**Expected results**
X and Y dimension should change leaving Z at its original scale.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-01-14:
Duplicate of https://github.com/Ultimaker/Cura/issues/4136.

closed 2020-01-14 by Ghostkeeper

g2721, 2020-02-18:
Having the some Problem will the be a fix some time around?
MacOS; Cura 4.4.1

Ghostkeeper, 2020-02-25:
I hope so. It's a very serious front-end issue. But it's hard to tell! We've pushed this along for some years now.