Ultimaker/Cura #6973: Build volume temperature can not be set

maukcc, 2020-01-14:
Cura PPA master
When setting the Build volume temperature manually, it automatically jumps back to the minimum value.
Bug or feature?
If you select another profile it does show the changed value at the customized settings.

mahtDFR, 2020-01-14:
Can you give me some specific reproduction steps here?

maukcc, 2020-01-14:
The build volume temperature is at 20 degrees, manually change it to 30 and press enter. 30 remains. When the cursor goes out of the build volume temperature field it changes back to 20. 
The gcode file will read M141 S20.
A change profile dialog box will show a customized build volume temperature of 30.

mahtDFR, 2020-01-14:
Not seeing a build volume temperature. Which printer profile are you using for this? Would you be able to share it with me? It will make it easier to reproduce the issue.
Also, which step of the PPA is it on?

Ghostkeeper, 2020-01-14:
Yeah I could reproduce this (with the additional steps of installing the Printer Settings plug-in first and checking the "Has Build Volume Temperature Stabilization" setting then). It's not in the PPA.

It's probably related to the bug we found yesterday with the prime tower setting not working, and the reports from @smartavionics in a certain commit that I can't find back any more. This is a critical.

closed 2024-12-07 by github-actions[bot]