Ultimaker/Cura #6975: [4.4.1] Materials actions don't update until closing the settings

Liger0, 2020-01-14:
**Application version**
4.4.1

**Platform**
Win10 x64

**Printer**
Any

**Reproduction steps**
1. Edit materials

**Screenshot(s)**
https://www.youtube.com/watch?v=r3Z1AZVdCsI

**Actual results**
Material aren't updated. Settings restart is required

**Expected results**
No setting restart required

fieldOfView, 2020-01-15:
Do you have the Material Settings plugin installed? I can't tell from your reproduction steps or your video. If so, could you try with that plugin disabled?

Ghostkeeper, 2020-01-15:
I couldn't reproduce this issue in current Master. Maybe it's only occurring with the Material Settings plug-in. Maybe it's been fixed since?

Do you have a log file available for us? It could indicate some error there halfway through deleting the material that prevents the screen from updating.

Liger0, 2020-01-15:
I don't have the material plugin.
[cura.zip](https://github.com/Ultimaker/Cura/files/4067326/cura.zip)

Ghostkeeper, 2020-01-17:
Maybe this is indicating the problem?

    2020-01-15 00:20:30,990 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [58]: Could not get metadata of container Unknown because it was not found.

Sadly it doesn't put a trace there to see if this came from the material manager.

We're still unable to reproduce the problem on this end. Adding or removing a material is immediately reflected in the list of materials.

neilpa, 2020-01-24:
I can repro this on macOS as well, version 4.4.1 installed via Homebrew with similar logging when I open the materials panel

```
2020-01-24 09:44:36,475 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [615]: Could not find plugin global_stack
2020-01-24 09:44:36,482 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [615]: Could not find plugin extruder_stack
2020-01-24 09:44:36,493 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [58]: Could not get metadata of container Unknown because it was not found.
2020-01-24 09:44:36,636 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [58]: Could not get metadata of container Unknown because it was not found.
2020-01-24 09:44:36,657 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/Preferences/Materials/MaterialsDetailsPanel.qml:91: TypeError: Cannot read property 'container_node' of null
2020-01-24 09:44:36,671 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/Preferences/Materials/MaterialsPage.qml:271:13: QML MaterialsList: Binding loop detected for property "height"
2020-01-24 09:44:36,675 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/Preferences/Materials/MaterialsList.qml:136:5: QML Column: Binding loop detected for property "height"
```

Happy to provide more info or try a different build if that helps

calinb7, 2020-02-08:
I have this problem on Cura 4.4.1 Linux AppImage too. (MX-18 Linux / Debian-based system). It does not occur with my last version (Cura 4.2.1). So there you have it. This bug wins the triple crown (occurs on Windows, MacOS, AND Linux)! Don't know why you guys can't smash it.

nallath, 2020-02-10:
> Don't know why you guys can't smash it.

Because we can't reproduce it. If we can't reproduce it, there is no way to figure out if any changes that we make actually fix the issue. It's a bit like trying to repair a car, but you need to do it over the phone and the other side only as a single wrench.

calinb7, 2020-02-11:
I could author a bootable "live" (with persistence) MX-18 Linux .iso file for you. It will already have the appimage on it. If you write it to a bootable USB and either run it live or even install it.

The appimage seems very sluggish (more than usual). This may be a case of such poor performance that the automatic refresh is just so slow that most of us conclude that it isn't happening and force it to occur by closing the settings after we lose patience. You devs probably have much faster machines than I!

calinb7, 2020-02-11:
I just checked to see if the refresh is simply very sluggish but no--refresh is simply absent, One can select another option to manage on the left side window and then switch back to Materials to refresh the Materials screen too. You don't have to close the entire window.

Most MX-18 Linux devs do their work on live / persistent systems (saves all changes). You could probably install enough of your tools on a live MX-18 system that I provide to you to figure this out. Presumably the fix would apply to Mac OS and Windows too. Please let me know if you would like me to clone my system to a live .iso file. MX has just about the most powerful "live" features around! More info here:
https://mxlinux.org/wiki/system/create-a-live-usb-w-persist-from-a-windows-desktop/

mahtDFR, 2020-02-13:
Just tried this in 4.5 beta Windows 10. No Materials settings plugins enabled.
Couldn't reproduce the create duplicate behaviour, the duplicate appears (with about 1s of delay).

closed 2020-06-19 by no-response[bot]