Ultimaker/Cura #6976: Windows Security Update bricked Cura

Firedrops, 2020-01-15:
**Application version**
4.4.1
**Platform**
Windows 10 64-bit

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Update Windows. KB4532938, and KB4528760

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Blank white screen on start up. 

**Expected results**
Normal view of a 2D grid and UI elements

**Project file**
none

**Log file**
```
Thread 0x00004228 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001a6c (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "E:\Ultimaker Cura 4.4\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004304 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001af0 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000022ec (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1583 in wait
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1167 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001ac4 (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1583 in wait
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1031 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000007d4 (most recent call first):
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 960 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap
[... 118 lines cut ...]
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x000009fc (most recent call first):
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 306 in _logInfoWidget
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 156 in _createDialog
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 77 in __init__
  File "X:\4.4-exe\build\inst\bin\cura_app.py", line 111 in exceptHook
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 341 in exec_
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 804 in run
  File "X:\4.4-exe\build\inst\bin\cura_app.py", line 180 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
```

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-01-15:
Maybe you could try updating your graphics drivers as well? It could be an incompatibility between your graphics driver and Windows.

Firedrops, 2020-01-16:
Thanks, that was indeed a fix.

Bloody 5700 drivers...

Firedrops, 2020-01-16:
aaaaaaaaaaand it's back. Successfully opened it once then it's blank white screen again. Not sure if it's the same errors:
Error traceback:
```
Traceback (most recent call last):
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 183, in mouseReleaseEvent
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtMouseDevice.py", line 36, in handleEvent
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Controller.py", line 344, in event
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 1806, in _onContextMenuRequested
AttributeError: 'NoneType' object has no attribute 'getIdAtPosition'
```
Logs:
```
Thread 0x000011a8 (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1583 in wait
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1167 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002268 (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1583 in wait
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1031 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004648 (most recent call first):
  File "C:\Users\Larry\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 960 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000e10 (most recent call first):
  File "E:\Ultimaker Cura 4.4\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000044dc (most recent call first):
  File "E:\Ultimaker Cura 4.4\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000016f0 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004a8c (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "E:\Ultimaker Cura 4.4\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002284 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000f4c (most recent call first):
[... 106 lines cut ...]
  File "threading.py", line 424 in acquire
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00002fac (most recent call first):
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 306 in _logInfoWidget
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 156 in _createDialog
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 77 in __init__
  File "X:\4.4-exe\build\inst\bin\cura_app.py", line 111 in exceptHook
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 341 in exec_
  File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 804 in run
  File "X:\4.4-exe\build\inst\bin\cura_app.py", line 180 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
```

Twigho, 2020-01-16:
Hello,
On the off chance, do you have the "controlled folder acess" option activated?

Firedrops, 2020-01-16:
I don't think so, I can't find it in Windows Security options nor in regedit, which are places to check according to guides.

Twigho, 2020-01-16:
This is found in Windows security/virus&threat protection/virus&threat protection setting/manage settings and below Tamper protection you have Controlled folder access/manage controlled folder access.
But perhaps it would be easier to look directly in the history. (windows security / virus & threat protection / protection history. If you have some messages like this :
[image]

Firedrops, 2020-01-17:
I really do not have these options in my Virus & Protection settings, I wonder if it's because I'm using a 3rd party antivirus?

Either way, I checked both Windows' and my antivirus' events history, and there's nothing related to Cura in there.

The only workaround I've found so far is to enable both Windows 7 compatibility and Administrator mode (Win 8 won't work). Occasionally I'll still be unable to run Cura, in which case I have to turn off compatibility mode, try to run, let it crash, and turn it back on again... Very strange but it works?

Twigho, 2020-01-17:
Hello,
Windows security blocking would have been easier.

As Ghostkeeper mentioned, this looks like a compatibility issue with windows and cg drivers. You mentioned a Radeon 5700, right? This card is rather reputed to be a bit capricious. Are you using an automatic overcloking mode? A fairly powerful PSU? Or is your RAM in XMP mode?

PS: I have to tell you that I am not from the dev team nor affiliated with.

Ghostkeeper, 2020-01-17:
Maybe we need to look at a different approach.

It's weird, that error that you're getting. The code it's mentioning should ONLY be executed when right clicking on the screen. I can't reproduce it, but I can theorise why it's giving an error. Perhaps the rendering hasn't been initialised yet so there is no selection render pass yet.

I've made a fix now so that this error shouldn't occur any more, but since I can't reproduce the original issue I can't really verify if it fixed it for you. Perhaps it would now crash in another place...

Firedrops, 2020-01-17:
@Twigho Yes I'm aware of the various driver issues with the 5700... ugh. Anyway, no I'm not overclocking the GPU, my PSU is 750w gold from EVGA, and RAM tuned using 1usmus' settings, so even better than XMP. Might have to report this to AMD directly? 

@Ghostkeeper Ah I see... Truth is the blank screen stays blank and will not generate any error logs until I right click on it, so I assumed the right click was the "final straw" and the error log was related to the blank screen issue. I'll see how it goes with the next Cura update.

Ghostkeeper, 2020-01-17:
Oh in that case I probably didn't fix the issue. I fixed an issue where it would crash if you right-clicked before the first render takes place. In other cases this is very hard to do because the render takes 10ms or so, just in your case it was crashing because for some reason it's not completing that render.

Twigho, 2020-01-17:
No, I don't think it's necessary to report to AMD. It just looks like Cura is waiting for something to finish loading. (A similar thing happened to me when Windows security blocked a part of Cura and was waiting for an authorization. Which was my first suggestion)
Out of curiosity, when you were talking about Window 7 compatibility and admin mode. You apply it only on cura.exe or also on curaengine.exe

Ghostkeeper, 2020-01-17:
I don't expect there will be any difference with CuraEngine.exe since it doesn't call any native Windows functionality. Try with cura.exe only.

Twigho, 2020-01-17:
@Ghostkeeper Does Cura require a minimum of virtual memory ? ( For example, 60% of the vram)

Ghostkeeper, 2020-01-17:
I think on the Ultimaker website we say that you need at least 4GB system memory. I know that with typical usage Cura takes about 15% of my VRAM (2GB) and like 200-400MB of actual memory.

Firedrops, 2020-01-18:
@Twigho the compatibility workaround is done on Cura.exe. Haven't tried anything with CuraEngine.exe. I'm sure I've clicked on "Troubleshoot Compatibility" option a few times while experimenting with Win8 compatibility or re-toggling the Win7 compatibility, but itdoesn't seem to do anything?

Don't think RAM is an issue, I've got 16 GB and very few background processes. Is there a way to be sure?

Twigho, 2020-01-18:
Do not worry, a RAM problem often leads to a crash of windows (or bluescreen).
Regarding virtual memory, I was just wondering because I have already encountered some rendering engines that require a minimum. (More than windows gives with the automatic option). But I tested with 800MB (the minimum recommended for windows) and Cura works correctly.
"Troubleshoot compatibility" doesn't work often ... It's a bit like waiting for a miracle.

KB4532938 is an update for .NET framework 3.5 and 4.8 .. And the other "improve security when using input devices such as a mouse, keyboard, or stylus." Perhaps the direction taken with the right click is not bad.
And if you're trying to change your port mouse or reinstall it, if it requires proprietary drivers. It's a bit silly but you never know with windows.

The rest exceeds my skills, remains to wait for the result of the dev investigation.

Ghostkeeper, 2020-01-20:
We haven't seen this problem yet on any of our own test computers, but we haven't been able to properly test this since the computers we have don't get those update numbers (KB4532938 and KB4528760).

Can you try running Cura in compatibility mode as Windows 8? We have bug reports with similar symptoms where this was a workaround.

Firedrops, 2020-01-21:
Doesn't look like mouse drivers are helping, my mouse actually stores settings on-board and the "driver" is only manually run to change those, not an auto-start in the background thing.

I've tried Windows 8 compatibility first, it didn't work for me, that's why I moved further back to Windows 7. It's still working for me like this with a re-toggle once every day or 2.

Ghostkeeper, 2020-01-21:
Another thing that's mentioned in other threads is that the Intel graphics drivers had a bug that was affecting Cura where resetting the OpenGL context would freeze the application. They released an updated video driver for that last week. So if you're using Intel as your GPU (embedded in the CPU), you can try updating video drivers as well.

Timocop, 2020-01-21:
Just throwing this in here. I have those updates installed. Cura works fine for me. I have WIndows Defender as main AntiVirus.
[image]

commit 0e5654e4 (2020-01-17): Guard against selection pass not existing yet

It could happen that the selection pass is not initialised because you're right clicking on the screen before the first render has happened.

Hopefully this fixes #6976.
closed 2020-06-19 by no-response[bot]