Ultimaker/Cura #6977: [master] Quick question - how does getSceneLock() actually stop scene from being rendered?

smartavionics, 2020-01-15:
I suspect that the scene locking in camera tool is ineffective, can someone please explain how acquiring the scene lock stops the scene from being rendered? I can't find in the source where the rendering acquires or tests the lock?

Please close this issue after answering.

Thanks.

nallath, 2020-01-15:
I don't think the scene lock is even needed. It's probably one of those things that was needed at some point in time but got lost during the refactoring.

smartavionics, 2020-01-15:
Thanks for the quick response. Well, I'm seeing a screen jitter problem here that looks like the scene is getting rendered between these two lines in _rotateCamera()...
```
        camera.setPosition(n)
        camera.lookAt(self._origin)
```
Later today, I will try and stop the rendering between those lines and see if the jitter goes away.

nallath, 2020-01-15:
If you change the _render function in MainWindow to, it should use the lock
```python3
def _render(self):
    with Application.getInstance().getController().getScene().getSceneLock():
        if self._full_render_required:
            renderer = self._app.getRenderer()
            view = self._app.getController().getActiveView()
            renderer.beginRendering()
            view.beginRendering()
            renderer.render()
            view.endRendering()
            renderer.endRendering()
            self._full_render_required = False
            self.renderCompleted.emit()
        else:
            self._app.getRenderer().reRenderLast()
```

Would it be possible for you to make a video of the jitter? I've never noticed any jitter, so I'm curious what it is that you're experiencing.

smartavionics, 2020-01-15:
Thanks for that suggested fix. I haven't tried it but I suspect it would not have helped. I have since discovered that the jitter occurs on WIndows when _rotateCamera() is executed not on the main thread. Linux doesn't care but Windows seems to do the render for the setPosition() before doing the render for the lookAt(). So my simple solution is to use signals to ensure that the call to _rotateCamera() is executed on the main thread. I guess you should get rid of the scene lock sometime as it's not doing anything useful.

closed 2020-01-15 by smartavionics